makerbase-mks/MKS-Robin-E3-E3D #1: part cooling fan is always open

SFRSKZ, 2020-04-09:
im using pa8 pin for part cooling fan but fan is always working. not responding to adjustments or gcodes.

makerbase-mks, 2020-04-10:
Hi!
I want to confirm the following points:
First, your firmware is what version?
Secondly, when the motherboard is powered on, will the FAN controlled by PA8 flash or keep on?

SFRSKZ, 2020-04-10:
marlin 2.0.1. i used your files, and changed some settings but there was nothing about fan.

SFRSKZ, 2020-04-10:
[image]

red light is always on, not blinking.

makerbase-mks, 2020-04-10:
Have you received the motherboard, is this the case with power on? If it is always on, it means that the transistor controlling the FAN has been broken down, which will cause the power to be on all the time without blinking.

closed 2020-05-13 by makerbase-mks

jvictor23br, 2020-08-19:
Mine has the same issue, the red light is allways on, but the cooler is always off. Just received like that. What can be done?

danbrianwhite, 2020-11-11:
I have one of these too where part cooling fan is always on. The hotend fan is connected to the "fan" input.

Is this a software bug?