makerbase-mks/MKS-Robin-E3-E3D #2: I can't update .bin file

JordiKoki, 2020-04-20:
I am trying to update my new MKS Robin E3D board, but when I add the file Robin_e3.bin in my SD, the screen does not execute the update.

makerbase-mks, 2020-04-20:
Hi! 
1. What screen are you using?
2. Robin E3 / E3D already have firmware. If you have not modified the firmware, you can use it directly on the Ender3 machine without upgrading
3. Robin E3 / E3D supports LCD12864 to display the firmware update progress

JordiKoki, 2020-04-20:
I use a TFT35 V1.0 (with Firmware 1.0.5) in my CR10s, but doesn't works anything (hotend, bed, moves, fans...)

makerbase-mks, 2020-04-24:
Hi! You can refer to it and link is : https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Upgrade-with-MKS-TFT-Series

closed 2020-05-13 by makerbase-mks

spooky79, 2021-04-21:
I have the same problem, it comes to 100% and nothing is happening, I'm using visual studio code, any solution?

aspaixao, 2021-07-16:
I have the same with my e3d and I using a CoreXY and configuration to ender 6

sergiofagundes, 2021-09-16:
FIXED 100% working:

 the board if using a 4GB sdcard,
I tried with 8GB/16GB/32GB and none of them worked.
So I tried a 4GB sdcard and it worked.

I used the Sdcard on the LCD sdcard slot, but I think it should work also in the mks robin board sdcard slot also with only 4GB sdcard.