makerbase-mks/MKS-Robin-E3-E3D #3: question: in where save eeprom settings

sl1pkn07, 2020-04-23:
Hi

in this board, in where is stored the eeprom settings?

greetings

makerbase-mks, 2020-04-24:
Hi! The processing of this part of EEPROM, marlin may write to STM32 flash using an analog method.

sl1pkn07, 2020-04-24:
i see in other SMT32F103(RCT6/RET6) based boards this

~~~~
#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE     (0x800U) // 2KB
#define EEPROM_START_ADDRESS (0x8000000UL + (STM32_FLASH_SIZE) * 1024UL - (EEPROM_PAGE_SIZE) * 2UL)
#undef E2END
#define E2END                (EEPROM_PAGE_SIZE - 1) // 2KB
~~~

but not in the Robin E3/E3D

this is missing for the Robin E3/E3D boards?

(i have the E3D in route)

greetings

sl1pkn07, 2020-05-15:
Hi again. one question, with "analog method" you mean the board have a dedicated eeprom chip?

greetings

DocYester, 2020-05-17:
saving settings to flash with the e3d would be nice - actually i store them on sdcard by emulation
but that will definitly fail at some time - so allways backup the settings

what is needed to set it up to use the flash ?

sl1pkn07, 2020-05-17:
in other STM32F1 boards, the snipplet i posted above

DocYester, 2020-05-17:
after rebooting 2 times and saving defaults - works 👍

sl1pkn07, 2020-05-17:
oh! I forgot to add to add this in the LD script

~~~
rom (rx)  : ORIGIN = 0x08007000, LENGTH = 256K - 28K - 4K
~~~

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/ldscripts/mks_robin_e3.ld

because the eeprom is stored in the last 4K in the flash

DocYester, 2020-05-19:
LD Script fot the E3D should be :
`rom (rx)  : ORIGIN = 0x08005000, LENGTH = 256K - 20K - 4K`

Works like a charm now

sl1pkn07, 2020-06-08:
implementes in upstream

https://github.com/MarlinFirmware/Marlin/commit/19024306fc834fbead89e749d64d352194befa84

closing

closed 2020-06-08 by sl1pkn07