makerbase-mks/MKS-Robin-E3-E3D #4: Can not Open build_flags.py

Frederic-SABU, 2020-05-01:
I can not compile a Marlin-2 for a ROBIN E3D.
I am using VS studio and Platformio.
My config is working fine for my other MKS boards : SGEN_L and MKS_BASE.
Error message is :
D:\Python38\python.exe: can't open file 'Marlin/src/HAL/HAL_STM32F1/build_flags.py': [Errno 2] No such file or directory
Thks for any help.

makerbase-mks, 2020-05-04:
What you set in the platformio.ini file is: mks_robin_e3d?

Frederic-SABU, 2020-05-04:
Yes, it is but the problem was coming from one of he marlin file. I 
reused a Marlin 2 from my SGEN_L cards and it is not compatible.

Thk a lot for following up.

Rgds,
Frd

Frederic-SABU, 2020-05-04:
No sorry it is :

mks_robin_e3

as there is no e3d environment.

Rgds,
Frd

closed 2020-05-27 by makerbase-mks