makerbase-mks/MKS-Robin-E3-E3D #8: Question : UART Connection

DocYester, 2020-05-19:
I have serious connection issues when i connect the e3d board via usb to a raspberry 3b+ with octoprint
tried diffrent cables - no change
i havent had theese issues before with the same pi/cable and the ender3 1.1.5 board
so i want to connect the e3d by uart to the pi
which pins do i need to connect ?
can i use the aux1 rx/tx/gnd ?
do i need a converter from 5v to 3.3v ?

makerbase-mks, 2020-05-20:
Hi! You don't need to connect power, but they need to be powered separately.

DocYester, 2020-05-20:
did it ... but cant connect at all
how to setup the firmware to enable the uart port for host communication ?
tried it with
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
```
and
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
```
both failed - tried to set the baudrate from  115200 to 250000 and 500000 - failed as well

makerbase-mks, 2020-05-21:
AUX-1 rx tx use STM32 USART1, so #define SERIAL_PORT_2 0;
USB is same as AUX-1, you can test connect to Host computer.

DocYester, 2020-05-22:
Not really ...

```
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:101:6: error: #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."

     #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:101:6: error: #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
     #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:101:6: error: #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
     #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:101:6: error: #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
     #error "SERIAL_PORT_2 cannot be 0. (Port 0 does not exist.) Please update your configuration."
      ^~~~~
*** [.pio\build\mks_robin_e3\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\mks_robin_e3\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\mks_robin_e3\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mks_robin_e3\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
========================= [FAILED] Took 5.80 seconds =========================
```

Trying to compile with Marlin bugfix branch 2.0.5.3

makerbase-mks, 2020-05-25:
Hi! Sorry, I corrected my mistake. Should #define SERIAL_PORT_2 1

DocYester, 2020-05-26:
After some testing and printing the only solution was to lower the baud rate to 115200 for a stable communication - uart or usb

closed 2020-05-26 by DocYester