makerbase-mks/MKS-Robin-E3-E3D #9: what problem ?

dct1122, 2020-05-20:
my use ronbin e3 broad 
why my extruder motor is noise ( mosquito sound ) ? 
i was change other motor , it not fix

makerbase-mks, 2020-05-20:
What driver do you use?

dct1122, 2020-05-20:
tmc 2209

makerbase-mks, 2020-05-21:
You can try to adjust the current to achieve the best condition of the motor.

DocYester, 2020-05-24:
my fans are louder than the stepper motors @40mm/s real print speed (slic3r , not cura !)
stepper motors are the original from the ender3 that it was before

```
Stepper driver current:
M906 X600 Y600 Z700 E800

```
```
Driver stepping mode:
M569 S1 X Y Z E

```
```
Maximum feedrates (units/s):
M203 X300.00 Y300.00 Z5.00 E25.00
Maximum Acceleration (units/s2):
M201 X1800.00 Y1800.00 Z100.00 E5000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1800.00 R3000.00 T2400.00

```

mks-viva, 2020-10-14:
Has this problem been solved?

mks-viva, 2020-11-17:
I will close this issues temporarily.

closed 2020-11-17 by mks-viva