makerbase-mks/MKS-Robin-E3-E3D #11: Compile error

vgege86, 2020-05-26:
I am trying to compile the newly downloaded version and it gives me the following error:

Compiling .pio\build\mks_robin_e3\libc6f\USBComposite\Keyboard.cpp.o
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
                 from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
In file included from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:2:0,
                 from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
In file included from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:0:
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_e3\libb69\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
In file included from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
                 from .pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\HAL_STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
Marlin\src\HAL\HAL_STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
     if (_dataOrder == SPI_BITORDER_LSBFIRST) {
                       ^~~~~~~~~~~~~~~~~~~~~
In file included from .pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.h:55:0,
                 from .pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\mks_robin_e3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: note: suggested alternative: 'SPI_CR1_LSBFIRST'
     if (_dataOrder == SPI_BITORDER_LSBFIRST) {
[... 24 lines cut ...]
                                ^~~~
*** [.pio\build\mks_robin_e3\libb69\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1
In file included from c:\users\vgege\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\inttypes.h:22:0,
                 from C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:4,
                 from C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29,
                 from C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
                 from C:\Users\vgege\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.h:50,
                 from .pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' uses the default argument for parameter 3, which is not yet defined
   Adafruit_SPIDevice spi_dev = NULL;
                                ^
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t, int8_t, int8_t, int8_t)':
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp:38:76: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BitOrder, uint8_t)' uses the default argument for parameter 6, which is not yet defined
   spi_dev = Adafruit_SPIDevice(spi_cs, spi_clk, spi_miso, spi_mosi, 1000000);
                                                                            ^
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t)':
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
       Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\mks_robin_e3\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
       Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
*** [.pio\build\mks_robin_e3\libc7b\Adafruit_MAX31865\Adafruit_MAX31865.cpp.o] Error 1

erdorath, 2020-05-26:
I can confirm this issue, It seems to be related to this Marlin issue:

https://github.com/MarlinFirmware/Marlin/issues/18081

Changing the library version to 
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip

Seems to solve the problem! Maybe update this in the configuration files?

makerbase-mks, 2020-05-27:
@vgege86 You can try his @erdorath  method.

m-sja, 2020-05-29:
I can confirm changing Line 34 in platformio.ini solves the compile error.

[image]

Thanks @erdorath

makerbase-mks, 2020-05-30:
Can you provide your configuration file and .ini file .

DocYester, 2020-05-30:
hmm... looks like not being an issue ...
https://github.com/MarlinFirmware/Marlin/issues/18081#issuecomment-635688767

gromgit, 2020-06-01:
Just tried the fix mentioned in https://github.com/MarlinFirmware/Marlin/pull/18089/files (linked to in the comment mentioned by @DocYester), and it builds OK. Basically, in `platform.ini`, change:
```
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
```
to:
```
Adafruit MAX31865 library@>=1.1,<1.2
```
(note the spaces).

erdorath, 2020-06-01:
@makerbase-mks Could you reflect this fix @gromgit posted to the firmware version on your github? A lot of people are having trouble compiling the firmware because of this! It's an easy fix but it would be better if the firmware compiled out of the box! 
Thank you :)

gromgit, 2020-06-02:
@makerbase-mks To make things easier, I've backported the fix I mentioned as PR #14. Please merge both that and @DocYester's earlier PR, otherwise everyone else trying to rebuild their firmware will run into the same issues. Thanks much!

makerbase-mks, 2020-06-03:
OK！

closed 2020-06-03 by makerbase-mks