makerbase-mks/MKS-Robin-E3-E3D #12: Error compiling with ftdi_eve_touch_ui part.

n0th1n6, 2020-05-28:
Hi, I have been pulling my hair due to this issue. I have compiled some 32bit firmware, even for the ESP32 board. This one I cant figure out what is the problem. Please see log below

```
Compiling .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\standard_char_set.cpp.o
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\event_loop.cpp:23:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\event_loop.cpp.d: No such file
or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\event_loop.cpp.o] Error 1
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\screen_types.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\screen_types.cpp.d: No such file or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\screen_types.cpp.o] Error 1
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\sound_player.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\sound_player.cpp.d: No such file or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\sound_player.cpp.o] Error 1
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\tiny_timer.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\tiny_timer.cpp.d: No such file
or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\tiny_timer.cpp.o] Error 1
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\text_box.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\../../compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\text_box.cpp.d: No such file or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\text_box.cpp.o] Error 1
In file included from c:\mks\marlin2.0 for ender3\marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp:22:
c:\mks\marlin2.0 for ender3\marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp.d: No
such file or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp.o] Error 1
In file included from c:\mks\marlin2.0 for ender3\marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\config.h:24:0,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_size_t.cpp:22:
c:\mks\marlin2.0 for ender3\marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\compat.h:48:47: fatal error: opening dependency file .pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_size_t.cpp.d: No such file or directory
 typedef const __FlashStringHelper *progmem_str;
                                               ^
compilation terminated.
*** [.pio\build\mks_robin_e3d\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_size_t.cpp.o] Error 1
[... 10 lines cut ...]
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1381:
    _exec_main(parser, values)
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1344:
    _main(parser)
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1119:
    nodes = _build_targets(fs, options, targets, target_top)
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1318:
    jobs.run(postfunc = jobs_postfunc)
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Job.py", line 113:
    postfunc()
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1315:
    SCons.SConsign.write()
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\SConsign.py", line 117:
    syncmethod()
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 147:
    f = self._open(self._tmp_name, "wb", self._mode)
Exception ignored in: <function dblite.__del__ at 0x000001FD6D3322F0>
Traceback (most recent call last):
[... cut ...] n __del__
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 140, in close
  File "C:\Users\enolraf\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 147, in sync
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\mks\\marlin2.0 for Ender3\\.pio\\build\\mks_robin_e3d\\.sconsign37.db.tmp'
================================================= [FAILED] Took 293.88 seconds =================================================

```

makerbase-mks, 2020-05-30:
Hi! Can you provide your configuration file and .ini file .

n0th1n6, 2020-05-30:
I have made some changes already. I have deleted the whole extensible_ui directory and I was able to successfully build. Ill attach my config and ini files.
[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/4704901/Configuration.txt)

[platformio.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/4704902/platformio.txt)

makerbase-mks, 2020-06-03:
You needn't use #define CR10_STOCKDISPLAY
 you can use #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Because have not define pin in pins_MKS_ROBIN_E3D.h about CR10_STOCKDISPLAY display pin, so, you need enable #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

n0th1n6, 2020-06-03:
If I use the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, will it work with ender 3 LCD?

I have successfully built the firmware now after deleting the whole extensible_ui directory

mks-viva, 2020-10-14:
@n0th1n6 The firmware compile is OK.

mks-viva, 2020-11-17:
I will close this issues temporarily.

closed 2020-11-17 by mks-viva