makerbase-mks/MKS-Robin-E3-E3D #13: Error saving config in TFCard

KiZDGGit, 2020-05-30:
Hello, I'm just trying to save config with "stock firmware", but settings are not storing in SDCard not EEPROM.

How can I save config?

Thanks in advance.

DocYester, 2020-05-30:
Build Marlin from latest bugfix branch by yourself and see how to get the saving done to flash by emulation there -> 
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/3

KiZDGGit, 2020-05-30:
Thank you for your answer. Desisted to update using
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/marlin2.0%20for%20Ender3

With Arduino IDE, many errors:
ATENCIÓN: la librería U8glib pretende ejecutarse sobre arquitectura(s) avr, sam y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) STM32F1.
ATENCIÓN: la librería TMCStepper pretende ejecutarse sobre arquitectura(s) avr y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) STM32F1.
In file included from sketch/src/inc/MarlinConfig.h:38:0,
                 from sketch/src/Marlin.h:24,
                 from sketch/src/Marlin.cpp:31:
sketch/src/inc/SanityCheck.h:2084:4: error: #error "TMC2208 Software Serial is supported only on AVR, LPC1768, STM32F1 and STM32F4 platforms."
   #error "TMC2208 Software Serial is supported only on AVR, LPC1768, STM32F1 and STM32F4 platforms."

and with Platformio... some others (...and I don't like/understand how it works).

Is there any way to save settings somewhere with stock fw? I can't setup PID, extrusion...

Thanks!!

DocYester, 2020-05-31:
If you use the real bugfix branch -> https://github.com/MarlinFirmware/Marlin it works

I havent tested the stock firmware at all - so i cant tell you how if or where it could be saved at all

The only way else would be to put all together in a gcode startscript.
Most slicers support that.

KiZDGGit, 2020-05-31:
Updated with real bugfix branch!! motors rotated in the opposite direction, but now it's solved.

Thanks!

DocYester, 2020-05-31:
Look for
```
// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
```
in the Configuration.h

KiZDGGit, 2020-05-31:
> Look for
> 
> ```
> // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
> #define INVERT_X_DIR true
> #define INVERT_Y_DIR false
> #define INVERT_Z_DIR true
> ```
> 
> in the Configuration.h

Yes, that is the way I've solved. Now printing, but too high motor temperature, will check drivers settings again.

Thank you for your support.

DocYester, 2020-05-31:
TMC2209 and Ender3 Stock motors ?
Define the steppers in Configuration.h
```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE  TMC2209
```
Make them silent in Configuration_adv.h
```
#define STEALTHCHOP_XY
#define STEALTHCHOP_Z
#define STEALTHCHOP_E
```

Standard 2209-current settings in Marlin (800mA) is enough for them and they dont get too hot
Lowering them to 500-600-700 is possible but needs attention and testing if they still work fine