makerbase-mks/MKS-Robin-E3-E3D #15: Information about compiling Marlin

monkiki, 2020-06-02:
Hello, I need to compile Marlin to configure a BMG extrusor and already done with default Ender 3 Pro board, but not sure how to proceed with the Robin E3D board. For example, which boarch should I choose in Arduino IDE? Where can I get a default configuration for Marlin? Is the same as Ender 3 profile?

Thanks.

DocYester, 2020-06-02:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Build-and-Update-firmware

... try the default from mks here - but dont just copy all over - its an older version than actual marlin bugfix branch

makerbase-mks, 2020-06-03:
Hi! You can download new version marlin firmware and change it as use to Robin E3D board. Of course! You can download Robin E3-E3D path marlin and chang it.

closed 2020-06-03 by makerbase-mks

monkiki, 2020-06-03:
OK, thanks!