makerbase-mks/MKS-Robin-E3-E3D #16: KILLED. PRINTER HALTED

monkiki, 2020-06-03:
I was able to compile the provided Marlin from this repository but I got "KILLED. PRINTER HALTED" once the printer has the new firmware.

Printer: Ender 3 Pro
Board: MKS Robin E3D

n0th1n6, 2020-06-04:
What display are you using?

in my case, I use CR10_STOCKDISPLAY so I added this just after the #define CR10_STOCKDISPLAY. You can put it in Configuration_adv.h or even in board file. I guess it is best to put it in the board file as it could be board specific

Also disable psu control if you are not using one because it will assign a PIN and since there is no signal, it will still end up in this error

#if ENABLED(CR10_STOCKDISPLAY)
#define KILL_PIN -1
#endif

monkiki, 2020-06-04:
I use the Ender 3 stock display, should I use the same CR10_STOCKDISPLAY define? I will try to comment the KILL_PIN. I've also seen this workaround to avoid this problem but not sure if is a good practice. I've compiled Marlin for other printers (also the Ender 3 with stock board) and never got this error.

n0th1n6, 2020-06-04:
Ender 3 stock display, if you will notice is the same CR10_STOCKDISPLAY. It is safe because there is no kill button/pin in Ender 3. Unless you want to wire a kill button yourself but I believe you have to install the pullup/pulldown resistor.

monkiki, 2020-06-04:
Thanks for the info. I will try it.

makerbase-mks, 2020-06-04:
Hi! You download marlin firmware is Robin E3-E3D path? If you download it and use it, you need disable PSU_CONTROL

monkiki, 2020-06-04:
I disabled PSU_CONTROL but also needed to add `#define KILL_PIN -1` to **pins_MKS_ROBN_E3D.h** and got a lot of warning because KILL_PIN was redefined but now the error message dissapeared.

It would be really nice to download the source code hosted in this repository and compile it without any tweaks (at least with a stock Ender 3), because is a bit weird to get all these problems. I had to compile Marlin because I have a BMG.

n0th1n6, 2020-06-05:
I think you need it inside #ifdef to avoid those warnig

monkiki, 2020-06-05:
Same result.

Anyway my requirement is to be able to download the proper Marlin source code ready to be compiled for the MKS Robin E3D, or with well documented and working changes. 

But the best option should be submit these changes to the original Marlin project, so we can use a more recent version. I don't know why this board is not already included in Marlin like others.

DocYester, 2020-06-05:
It is - look in bugfix 2.0.5.3+ ... you just need the settings documented here in the "build & upload" section

monkiki, 2020-06-05:
In Marlin master I can see several ROBIN: LITE, NANO, MINI, PRO but no E3D :/

n0th1n6, 2020-06-06:
I am able to build the bugfix 2.0.5.3, still need some fixes here and there. The sample configuration that comes with bugfix has bedpidtemp enabled. That could be another surprise if you are not aware of it and has not tuned PID before.

I would like to commit some of my changes but since I am not a good C programmer, I avoided doing it. It can just mess up the source.

DocYester, 2020-06-06:
bugfix brnach ... the latest 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/boards.h
Line 306

> In Marlin master I can see several ROBIN: LITE, NANO, MINI, PRO but no E3D :/

bugfix branch ... not master
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/boards.h
Line 306

monkiki, 2020-06-06:
Thanks, I will try with this branch.

alcar21, 2020-06-09:
I have the same problem. Branch works?

Printer: Ender 5 Pro
Board: MKS Robin E3D

makerbase-mks, 2020-06-12:
You can download this bin file update and it have not PSU_CONTROL function
[image]

ToniJCustom, 2020-06-30:
Donwload Malin bugfix 2.0.x

Use configuration ender 3 from https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip

change archives from https://github.com/ToniJCustom/Config.-Marlin-2.0.x-Ender3-MKS-Robin-e3.git

if you have e3d board the onli change it in configuration.h

it is for stock machine configuration

DPSou, 2020-07-14:
> 
> 
> Donwload Malin bugfix 2.0.x
> 
> Use configuration ender 3 from https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip
> 
> change archives from https://github.com/ToniJCustom/Config.-Marlin-2.0.x-Ender3-MKS-Robin-e3.git
> 
> if you have e3d board the only change it in configuration.h
> 
> it is for stock machine configuration

I didn't get the second step. The only Ender 3 configuration files on the link are for the stock Ender 3 V1. (Configurations-bugfix-2.0.x\config\examples\Creality\Ender-3\CrealityV1).

ToniJCustom, 2020-07-26:
Yes, i change the language, i don't remenber sorry, the importante is the correct configuration of board and platformio