makerbase-mks/MKS-Robin-E3-E3D #17: Robin E3 does not work with 3D Touch via PB1 pin

chshkhr, 2020-06-05:
I spent the whole day trying to run 3D Touch with MKS Robin E3 using the tutorial https://www.youtube.com/watch?v=r2A0b7XQEaI without any success. Finally connected it on Z End Stop and it's OK now. Then I tried again to switch back to PB1 pin assured that 3D Touch is fine but nothing helps... 3D Touch work only via Z End Stop in my case

makerbase-mks, 2020-06-06:
You can refer to 3D Touch bltouch tutorial with LCD12864 and MKS Robin E3D SGen_L Gen_L :https://www.youtube.com/watch?v=r2A0b7XQEaI

chshkhr, 2020-06-06:
You recommend the same tutorial as I mention in my comment...

chshkhr, 2020-06-06:
By the way in both User Manuals for [E3D](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS-Robin-E3D-V1.0-User-Manual) and [E3](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS-Robin-E3D-V1.0-User-Manual) PIN diagrams reference on PB1 for ESTOP and PA2 for PROBE. But the motherboard contains just one connector for both Probe and Estop named "Probe/ESTOP". I tried `#define Z_MIN_PROBE_PIN PB1` and `#define Z_MIN_PROBE_PIN PA2`, `#define Z_MIN_PROBE_ENDSTOP_INVERTING true` and `#define Z_MIN_PROBE_ENDSTOP_INVERTING false`. But the result is the same: 3D Touch tested OK from the LCD but does not work as ENDSTOP

erezagami, 2020-06-06:
Hi,
Did you enable `#define USE_PROBE_FOR_Z_HOMING`
in the Configuration.h?

erezagami, 2020-06-06:
also did you set
`#define NUM_SERVOS 1 `
?

chshkhr, 2020-06-06:
I did everything from [the Tutorial](https://www.youtube.com/watch?v=r2A0b7XQEaI).
Including `#define NUM_SERVOS 1`  and `#define Z_MIN_PROBE_PIN PB1`
But `#define USE_PROBE_FOR_Z_HOMING` not mentioned in the Tutorial.
I searched throughout all source codes but `USE_PROBE_FOR_Z_HOMING` not used anywhere too...
So I think  `#define USE_PROBE_FOR_Z_HOMING` can't help

erezagami, 2020-06-06:
Are using the bugfix branch cause they add it recently it not at 2.0.5
3 release

chshkhr, 2020-06-06:
> 
> 
> Are using the bugfix branch cause they add it recently it not at 2.0.5
> 3 release

I'm using [MKS-Robin-E3-E3D](https://github.com/makerbase-mks/MKS-Robin-E3-E3D)...
What should I do to use sources from [Marlin main repository](https://github.com/MarlinFirmware/Marlin) (including bugfix branch) for MKS Robin E3?

erezagami, 2020-06-06:
I am using bugfix from marlin main repo. I have bltouch connected to the prob and it does work. I do not know when they sync last. We can check later

erezagami, 2020-06-06:
I do have an issue with the printer freezing but i do not think it firmware related

chshkhr, 2020-06-06:
Thank you! I will wait responce from makerbase-mks

chshkhr, 2020-06-07:
With sources from Marlin main repo [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) 3D Touch connected to the prob does work... Thanks to [erezagami ](https://github.com/erezagami)!

anilcancakir, 2020-07-24:
> With sources from Marlin main repo [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) 3D Touch connected to the prob does work... Thanks to [erezagami ](https://github.com/erezagami)!

I couldn't run the X & Y endstops with bugfix-2.0.x marlin version. Can you share with me?

erezagami, 2020-07-25:
you can look at my branch https://github.com/erezagami/Marlin/tree/2.0.x-bf-makerbase-256k
Note that there are some customization for my printer (bed size, kpi tuning, bltouch) etc...
so you probably do not want to use it as is.

tassiolucas, 2020-09-12:
I own an MKS Robin E3 and have tested all possible firmware. Both the MKS-Robin-E3-E3D-master (which in turn does not have the buldroot working) and the one marked with release tag and both the PB1 port does not work. - I also tested the newer versions of [MarlinFirmware / Marlin](https://github.com/MarlinFirmware/Marlin) and when flashing it doesn't even start Marlin. It is very sad to see this disorganization of code and documentation in which nothing makes sense. Frustrating!

makerbase-mks, 2020-09-14:
@tassiolucas Use to Ender3?

tassiolucas, 2020-09-16:
Yes, I use Creality Ender 3 PRO. I found Configuration.h and Configuration_adv.h for the printer. But even so, the BLTouch sensor recognizes, the extraction and retraction commands work, but the PB1 port in Bilinear calibration, at the beginning of printing it does not level with the table and I end up needing to use an Offset Z for each type of height on the table, as if I had manually calibrated it. I followed the tutorial posted by you and I did not get the sensor to fully function. Follow the wiki: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH

Another strange thing that my sensor only passed the expansion and retraction test when I inverted the pins S G 5V by 5V G S; and the same thing for PB G from S G I had to change to G S.

Is this wiki really up to date?

makerbase-mks, 2020-09-17:
@tassiolucas 
Can you provide config to me?

tassiolucas, 2020-09-17:
Here the config file:

[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5240487/Configuration.zip)

makerbase-mks, 2020-09-18:
@tassiolucas 
You probe use Z-STOP? If is, you need sett it:
[image]

tassiolucas, 2020-09-18:
This makes no sense. At the beginning of the tutorial, you are asked to define Z_MIN_PROBE_PIN as PB1, which is where the sensor is connected. And at this point where you are indicating is to whom you are replacing the ENDSTOP of the Z axis. I do not replace and I am using the PB1 port to connect my 3D Touch. what is the point in commenting right after that line?

[image]

tassiolucas, 2020-09-18:
Other point. The sensor only works the mechanical part, when these pins have been inverted. From S G 5V to 5V G S (or Yellow -> Red -> Green, top to bottom in the photo on the plate)

[image]

Or else this is not the right scheme for Robin E3.

tassiolucas, 2020-09-18:
I tried to use this setup and it doesn't work. I need a setup that uses port PB1. With this, when trimming the ENDSTOP of the Z axis it stops, without getting the right level of 3D Touch.

makerbase-mks, 2020-09-19:
@tassiolucas 
We will configure the firmware that can use 3D TOUCH and upload it to Github.

tassiolucas, 2020-09-19:
This is not necessary. As the @chshkhr rightly said, Marlin does not recognize the PB1 as a 3D Touch sensor. I performed the second configuration indicated with 3D Touch on ENDSTOP of the Z axis and now the probe is working. I think you should update the configuration and because it is the only one that is present on the internet and its MKS Robin E3 cards do not have recognition on the port indicated as the default configuration which is PB1. Here are the configuration files that are the only ones that work, with the sensor connected to the ENDSTOP of the Z axis.

[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5249071/Configuration.zip)

mustafadh12, 2020-09-28:
hello, i have exactly same issue and ive started to have some dreams about configs, mks e3d and bl touch or maybe nightmares because of mks team!

mustafadh12, 2020-10-01:
can anyone share their files for ender 3 pro ?

mustafadh12, 2020-10-05:
> This is not necessary. As the chshkhr rightly said, Marlin does not recognize the PB1 as a 3D Touch sensor. I performed the second configuration indicated with 3D Touch on ENDSTOP of the Z axis and now the probe is working. I think you should update the configuration tutorial and because it is the only one that is present on the internet and its MKS Robin E3 cards do not have recognition on the port indicated as the default configuration which is PB1. Here are the configuration files that are the only ones that work, with the sensor connected to the ENDSTOP of the Z axis.
> 
> [Configuration.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5249071/Configuration.zip)

which version of marlin? im trying to compile but it says thats an old version and must be updated...

makerbase-mks, 2020-10-09:
We has upload it: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch

hieppo, 2020-11-13:
I'm having an extremely frustrating time trying to use 3d touch with the Robin e3 board and ender 3 pro. I have compiled the source mentioned just above and also tried the latest marlin release using the pb1 pins. Everything seems to be working fine until I try to print. The 25 points mesh is created and loaded and activated according to what the eeprom display. But everything print badly the nozzle does not seems to adjust and get close enough to the bed. Nothing stick well to the bed. I have tried various materials with different temp that worked perfectly when I manually level the bed. Do I really need to use the z limit pins. What's the difference between using pb1 and z pins. Mesh wad created through pb1 so how would switching to z helps with the print.

mks-viva, 2020-12-11:
I has upload 3d touch firmware to here, you can refer to it and you need set borad is BOARD_MKS_ROBIN_E3
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1

JonathanMortes, 2021-04-01:
I've got the same issue. I'm using a custom sensor that worked great for the original board and works with the new one (MKS E3P) but when trying to do auto leveling it attempts one point, detects it and then goes back home. I'm using the same Bilinear config with 9 points and 3x the probing as I did on the original board. Anyone can help?

MarioGK, 2021-04-19:
@JonathanMortes Having the same problem, tell me if you fixed it

JonathanMortes, 2021-04-19:
@MarioGK I did, I had to do some fancy stuff with the limits but I managed to make it work for my sensor and Ender 3. Let me know if you want the marlin ;)

jesterhead82, 2021-04-28:
I have an e3 running with Bltouch (w/o z Endststop), Sensor connected like in the picture 

[image]

Followed basically the Wiki entry
[
[Configuration.h.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6394090/Configuration.h.txt)
](url)

OpenZSD, 2021-05-02:
I'm having a similar issue. In my case I kept the z end stop. When touching the bed, the end retracts and the led lights but continues until the z end stop is reached. It goes to the first point in the bed leveling grid and then gives up. Any advice?

marcomro, 2021-05-04:
Não consigo fazer funcionar bltoutch de acordo com nenhum tutorial da própria fabricante. mks robin e3p v 2.0  alguem que possa me ajudar?

vinceshere, 2021-06-10:
> 
> 
> @MarioGK I did, I had to do some fancy stuff with the limits but I managed to make it work for my sensor and Ender 3. Let me know if you want the marlin ;)

Hello man! I know a lot of time passed by, but can you provide this Marlin please? I've already tried everything, but nothing works. I have a Ender 3 with a Robin E3 V1.0

DogsoldierNZ, 2021-08-24:
Did anyone find a fix for this ?

hieppo, 2021-08-24:
Those having a hard time with configuring BL/3D Touch on the Robin E3/E3D. This is what I notice different from what MKS documented.
/**
 * Enable this option for a probe connected to the Z-MIN pin.
 * The probe replaces the Z-MIN endstop and is used for Z homing.
 * (Automatically enables USE_PROBE_FOR_Z_HOMING.)
 */
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN ******NOTE DISABLE this if you want to use PB1 pin, with this disable, you can even unplug the Z limit switch without any issue******

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING  ******ENABLE this if you want to use PB1*******

/**
 * Z_MIN_PROBE_PIN
 *
 * Define this pin if the probe is not connected to Z_MIN_PIN.
 * If not defined the default pin for the selected MOTHERBOARD
 * will be used. Most of the time the default is what you want.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 *
 */
#define Z_MIN_PROBE_PIN PB1 // Pin 32 is the RAMPS default  *************Make sure this is enabled and PB1

/**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
 #define BLTOUCH ********Just in case you missed this, ENABLE it***************

DogsoldierNZ, 2021-08-24:
#define Z_MIN_PROBE_PIN PB1 // Pin 32 is the RAMPS default *************Make sure this is enabled and PB1
I did this in the Pins file and removed the Z_MAX_PIN and everything started to work. 

Your fix works perfectly thanks

santiagoluz, 2023-07-23:
Thanks @hieppo and @erezagami 
I was struggling with this issue. I'm using the probe connected to the slots in the board shown in the photo/diagram in the wiki article (https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH). It only worked correctly when I enabled `USE_PROBE_FOR_Z_HOMING`. I'm using Marlin 2.1.2 and BOARD_MKS_ROBIN_E3D.

**Configuration.h** section ended up like this:

```
/**
 * Enable this option for a probe connected to the Z-MIN pin.
 * The probe replaces the Z-MIN endstop and is used for Z homing.
 * (Automatically enables USE_PROBE_FOR_Z_HOMING.)
 */
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING 

/**
 * Z_MIN_PROBE_PIN
 *
 * Define this pin if the probe is not connected to Z_MIN_PIN.
 * If not defined the default pin for the selected MOTHERBOARD
 * will be used. Most of the time the default is what you want.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 */
#define Z_MIN_PROBE_PIN PB1
```

@makerbase-mks please update wiki *Firmware setting* section to include this instruction.