makerbase-mks/MKS-Robin-E3-E3D #18: Printer Halted at randomly when printing from usb

erezagami, 2020-06-06:
HI,
I have mks_robin_e3 with ender 3.
I am using the master branch of marlin 2.0.0-bugfix.
When using octprint over usb the printer halt, the screen become unresponsive and the serial connection is dropped and it not possible to reconnect unless restarting the printer.

The printer hang at different point every time, but when printing from sd card seems to complete successfully.

any suggestions ?

DocYester, 2020-06-06:
115200 Baud is actually the only suitable solution for me - got the same issue
Tried with UART-connect - no change

erezagami, 2020-06-06:
Thanks, I will try with 115200 Baud .

erezagami, 2020-06-07:
Setting the Baud rate to 115200 seems to solve  the freezing!!
3 prints so far all successful. 
Thanks DocYester!

Do we have any idea why the baud rate matter? I thought it emulates the serial?
Also i seems to have more stringing could that be related?

DocYester, 2020-06-07:
I would guess a bad SoftwareSerial libary
Just to get some numbers right : 115200 Baud = 14.kb/s
That should be more than enough to print even at real 60mm/s
When i slice a 3DBenchy @ 100% size with 0.12 layer and 0.5 line width in prusa slicer it's only 3.85MB gcode file - could be transferred in ~280 seconds (4:40) - but printing time (at 30mm/s) is roughly 3hrs - so i really dont care of 250k or 115k baud speed at all.

I havent seen any stringing getting worse by changing the baud rate yet.

makerbase-mks, 2020-06-08:
If you reduce the baud rate transmission speed can solve the problem, then congratulations. You can also try to set the baud rate to 250000, the cache instruction BUFSIZE is set to a larger in configuation_adv.h file, like 8 and so on

DocYester, 2020-06-08:
#define MAX_CMD_SIZE 384    // was 96
#define BUFSIZE 16           // was 4
#define RX_BUFFER_SIZE 256 // was disabled
Freezes still occour at 250k baud ...

makerbase-mks, 2020-06-11:
Regarding this problem, reducing the baud rate can be solved. So, I will temporarily close this question

closed 2020-06-12 by makerbase-mks