makerbase-mks/MKS-Robin-E3-E3D #19: Connect via usb with Octoprint and printer hanged.

avfoto, 2020-06-11:
Hi.
I have managed to connect my Robin E3D board mounted on my Ender 3 to my Octoprint system. For this I have to select "auto" in the baudrate.
After 10-15 minutes the printer hangs and stops printing (although it continues to heat)
What could be the problem?
Thank you.

erezagami, 2020-06-11:
try with 115200 baud rate.

avfoto, 2020-06-11:
I've already tried, it doesn't work.

makerbase-mks, 2020-06-12:
Hi!
1. Is the baud rate too fast, try using 115200
2. You can increase BUFFSIZE, such as 8

avfoto, 2020-06-12:
I've tryed that, but Octoprint only connect to the board when I specify "AUTO" on the baudrate

erezagami, 2020-06-12:
Do you compile marlin yourself? Did you set #define BAUDRATE 115200 
In marlin config?

avfoto, 2020-06-12:
Yes, I've compiled by myself, but my octoprint only connects with the printer when I select "AUTO" baudrate.

DocYester, 2020-06-20:
Dont even think about any help for this problem ...
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/18

DocYester, 2020-07-02:
Tried diffrent baud rates (56k/115k/250k) - even only usb1 mode (and all other stuff that is mentioned about "usb_serial_generic_read_bulk_callback - urb stopped: -32" errors) and 3 diffrent pi-hw-versions ... anyhere between 3 minutes and 7hrs the board just stucks - no more display updates, not responding to anything - only a hard reboot makes it responsible again
looks like to me that the mks board has the trouble-some chipset

mks-viva, 2020-10-15:
@avfoto Has this problem been solved?

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva