makerbase-mks/MKS-Robin-E3-E3D #20: Using 512K of flash & EEPROM Emulation

EvgeniyRU, 2020-06-13:
The board contains STM32F103RCT6 which has 512K flash, but current firmware settings using only 256K and not using EEPROM

To fix that it is needed to make some changes
[platformio.ini]
.....
platform      = ststm32
board         = genericSTM32F103RC
**board_upload.maximum_size=524288**
....

[mks_robin_e3.ld]
  **rom (rx)  : ORIGIN = 0x08005000, LENGTH = 512K - 24K**

[pins_MKS_ROBIN_E3.h, pins_MKS_ROBIN_E3D.h] at the end of file
**#undef STM32_FLASH_SIZE
#define STM32_FLASH_SIZE 512
#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE     uint16(0x800) // 2KB
#define EEPROM_START_ADDRESS uint32(0x08000000 + 512 * 1024 - 2 * EEPROM_PAGE_SIZE)
#undef E2END
#define E2END                (EEPROM_PAGE_SIZE - 1) // 2KB**

[Configuration.j]
uncomment #define EEPROM_SETTINGS

EvgeniyRU, 2020-06-13:
For only EEPROM emulation changes like that

[mks_robin_e3.ld]
**rom (rx) : ORIGIN = 0x08005000, LENGTH = 256K - 24K**

[pins_MKS_ROBIN_E3.h, pins_MKS_ROBIN_E3D.h] at the end of file
**#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE uint16(0x800) // 2KB
#define EEPROM_START_ADDRESS uint32(0x08000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE)
#undef E2END
#define E2END (EEPROM_PAGE_SIZE - 1) // 2KB**

[Configuration.h]
uncomment #define EEPROM_SETTINGS

makerbase-mks, 2020-08-05:
Hi! I has fix it and eeprom data storage

closed 2020-08-05 by makerbase-mks