makerbase-mks/MKS-Robin-E3-E3D #21: Cannot update firmware

haley0918, 2020-07-03:
I copied "Robin_e3.bin" from "MKS-Robin-E3-E3D-master\firmware\firmware bin for Ender3(TMC2209)" into sd card. Inserted the card to the board and powered up. But the lcd only shows "boot". After a while, removed sd card and reset. the lcd still shows "boot". 

How to solve this?

fadlio, 2020-07-03:
Maybe you forgot to include a LCD driver in configuration.h.
For my Ender 3, I uncommented the line with "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"

makerbase-mks, 2020-07-04:
@haley0918 You can try to download Robin E3-E3D path firmware bin file, and update it is normal？

haley0918, 2020-07-04:
@makerbase-mks I am trying with the given [Robin_e3.bin](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/firmware%20bin%20for%20Ender3(TMC2209)/Robin_e3.bin) file, not making from the Marlin source. Copied into sd card and insert into board then reset. LCD shows "boot". Reset it again and still same. Maybe file was not right in sd card, so I formatted my card and try again, no change, still only "boot".

@fadlio no I'm not making new firmware from source yet, just using the given pre-compiled ones. Will take note of that for when I compile from source.

makerbase-mks, 2020-07-04:
@haley0918 If you do not connect peripheral equipment, such as your motor, heating head... Can the motherboard start normally? You can just connect the LCD display and try to update the firmware?

haley0918, 2020-07-04:
@makerbase-mks Yes, I have nothing else connected. Only the motherboard and LCD, connected to usb power supply

haley0918, 2020-07-06:
No reply... does this mean my board is bricked? Nothing can be done to save it? I just got it not long ago, haven't begin using it yet.

@makerbase-mks

makerbase-mks, 2020-07-06:
Can you provide picture about board‘'s problem

haley0918, 2020-07-06:
[image]

makerbase-mks, 2020-07-06:
OK! I has view it. I need you to do the following: 
1. Connect your board to pronterface by USB and check it printing information;
2. Send M122 check information
[image]

haley0918, 2020-07-06:
@makerbase-mks here is a screenshot of M122.
[image]

haley0918, 2020-07-06:
@makerbase-mks  I tried a few more commands below.
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00

>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0

haley0918, 2020-07-06:
@makerbase-mks I tried a few more commands below.
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00

>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.1 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0

haley0918, 2020-07-06:
@makerbase-mks  there is something can reply on pronterface, but lcd still stuck at "boot".

haley0918, 2020-07-06:
I can see Marlin on lcd now. Maybe something went wrong when I first update and bootloader somehow "memorized" the card. Subsequent attempts with the same card would not update, no matter which firmware.

But changing to another sd card maybe cleared something in the bootloader and update could run again. Previous card could now also be used again for update.

I can now start compiling Marlin from official source to try. Making sure to include the right lcd driver in Configuration.h as @fadlio said.

Thanks.

closed 2020-07-06 by haley0918

jalius, 2020-09-03:
Hi. I have the same problem. Stuck at boot screen after installing Marlin without LCD driver. I can send commands to the printer with pronterface and get a reply. Can I reset the board to accept new bin file? I tried 2 different TF/SD cards, no luck.

>>> M122
SENDING:M122
echo:Unknown command: "M122"
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Sep  2 2020 20:39:14) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0

jalius, 2020-09-08:
Ok I just had to wait for something on the board to change/clear and the issue resolved itself. Now when I go to update the firmware with SD card it works fine. This time I made sure to enable the LCD on my ender 3 by un-commenting `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` in `configuration.h` as provided above. Works great, thank you.