makerbase-mks/MKS-Robin-E3-E3D #22: trying to add neopixel

jechs83, 2020-07-03:
i use the last  bin updated with neopixel fuctions added buy doesnt work.
i connnecta the stripe correctly but it doesnt work, i try to copile by my self using PA2 Pin but nothing 
Does anyone try it succescfully?

makerbase-mks, 2020-07-04:
You can refer to wiki about Neopixel light: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Upgrade-with-Neopixel-lights

closed 2020-08-05 by makerbase-mks

hieppo, 2021-06-07:
Hi,
Please update your wiki page to reflect the updated source code. I have read the wiki as indicated above but I could not find any reference mentioned in the page in the platformio.ini file. I am using the file in the MKS-Robin-E3-E3D path as suggested. Since the file looks different. I am not sure what is need to compile the neopixel.
Could you please use the updated files as reference to indicate the actual changes?

On another note, would anyone know if it is possible just to use the Neopixel pin as a switch to turn on/off a simple LED strip without any fancy brightness and such features?