makerbase-mks/MKS-Robin-E3-E3D #26: Conexion mks robin e3d y sensor inductivo

tolapoto, 2020-07-27:
Donde se debe de conectar correctamente el sensor inductivo en la plana robin e3d para no perder el final de carrera del eje z?

makerbase-mks, 2020-08-05:
Sorry, can you describe it more clearly?

hanger13, 2020-09-11:
Como conectar Sensor SN04 e como configurar no Marlin.

makerbase-mks, 2020-09-14:
1、SN04, you need use NPN type
[image]

tolapoto, 2020-09-14:
Gracias, lo he conseguido. Gracias a todos

closed 2020-09-14 by tolapoto

Emeale2021, 2023-03-31:
Ola, estou tentando instalar o senr indutivo na MKS Robin e3d, gostaria de saber qual pino conectar conectar

Emeale2021, 2023-03-31:
Hello, I am trying to install the inductive senr on the MKS Robin e3d, I would like to know which pin to connect to