makerbase-mks/MKS-Robin-E3-E3D #27: Robin_E3D-doesn't_update

TxasMan3D, 2020-08-04:
Robin_E3D+tft35 v1.0
Using VSCode to generate an updated Robin_e3.bin file. 
I save it in the MicroSD card and put it in the TF slot of the motherboard. 
When I repower the printer, board doesn't update with new file
Looks like the eeprom is disabled but, how can i enable it without comunication with the board?

makerbase-mks, 2020-08-05:
You can try to change TF card or format it. Can you provide your motherboard updating picture?

TxasMan3D, 2020-08-06:
I have used two different TF cards and my pc has no problems to read them. Yes, of course. I will send you by email the whole folder with the updated firmware.

TxasMan3D, 2020-08-06:
Due to the size of the folder i can’t send it so i send you the updated “firmware.bin” file that is located in .pio/build/mks_robin_e3. The name of the file was changed to “Robin_e3.bin”.

makerbase-mks, 2020-08-06:
No! You need copy Robin_e3.bin file to tf card, doesn't firmware.bin rename Robin_e3.bin 
You can download MKS Robin E3-E3D path Robin_e3.bin file update

TxasMan3D, 2020-08-06:
I had downloaded that one also but i have a bmg extruder fitted and i changed the steps in the firmware to get them saved in the motherboard, otherwise my problem has been fixed. I thought that the file that need to be copied was firmware.bin. I have try with the Robin_e3 one and looks like it has been read. Thank you!

closed 2020-08-06 by TxasMan3D