makerbase-mks/MKS-Robin-E3-E3D #30: bigthreetech tft24 não funciona na MKs Robin e3d. help

hanger13, 2020-08-24:
Tenho uma mks Robin e3d comprei um bigthreetech tft24 v1.2 . Porém não consigo por pra funcionar pela porta auxiliar 1 (aux1) apresentando o erro printer nota atached. Já mudei o baud rat para todos os valores e nada. Alguém na mesma situação?

makerbase-mks, 2020-08-25:
Hi! You need to do the following：
1、Connect TFT to Robin E3D, Must be correct
2、#define SERIAL_PORT -1 and #define SERIAL_PORT_2 1 ， #define BAUDRATE 250000
3、TFT baud rate must same as motherboard

hanger13, 2020-08-27:
tentei de tudo, nem via usb conecta.

makerbase-mks, 2020-08-28:
Usually only these three settings are needed to work normally. I think you need to confirm the TFT to AUX-1 wiring.

hanger13, 2020-08-28:
Confirmado. Os pinos do RX tx sao normalmente invertidos na mks Robin. Será que a placa está com problemas? Em curto Os pinos RX tx ?

makerbase-mks, 2020-08-28:
[image]

hanger13, 2020-08-28:
Essa é boa exata conexão que estou usando.mesmo assim não conecta de jeito nenhum. "Nenhuma impressora encontrar." Não movimenta nenhum eixo não habilita nenhuma saída de fan, hot end , mesa... Enfim não funciona.

makerbase-mks, 2020-08-28:
Can you provide your configuration.h file to me?

hanger13, 2020-09-05:
pinos rx tx em curto.vou adquirir outra placa

makerbase-mks, 2020-09-05:
You TFT？

hanger13, 2020-09-05:
Sim Meu TFT.

makerbase-mks, 2020-09-07:
I will colse this problem.

closed 2020-09-07 by makerbase-mks