makerbase-mks/MKS-Robin-E3-E3D #31: Requesting for Robin E3 firmware with bootloader.

livemeatz, 2020-08-27:
I have bricked my board due to using a corrupted TFcard.
Please provide a stock firmware for Robin E3 board which includes a **_bootloader_**,
so that i can reprogram the STM32 chipset via ST-Link V2 programmer card thru ST-Link Utility program.
Your help is greatly appreciated.

makerbase-mks, 2020-08-28:
Hi! You can refer to :https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/Bootloader
[image]

livemeatz, 2020-08-29:
I tried to run the given .hex firmware to no avail. Can you advice me on the programmer (may not be ST-Link V2) and re-flashing procedures..

makerbase-mks, 2020-08-31:
Use J-link tools and Jflash software tools

makerbase-mks, 2020-09-09:
Can you download it to your motherboard and go to run?

livemeatz, 2020-09-11:
[image]

Unable to load file. 
Could not execute task.
Could not program and verify target

makerbase-mks, 2020-09-14:
You need Unsecure chip and download it
[image]

livemeatz, 2020-09-15:
Successfully managed to program the stm32 chipset using STLink method (STlinkV2 doggle + software) and flash+verify the firmware using the provided mks_e3_hex file. 

However, the TF card issue still persist as the board does not recognise all the Robin_e3.bin files from the provided github "firmware" folder (the Robin_e3.bin does not change its filename and running M115 & M503 shows the unchanged default settings). I tried other micro SD cards, 2GB, 16GB, 64GB partitioned/formatted with FAT32, to no avail. 

Is there any possibility of me to update the board's firmware thru STLink (wired) method ?

makerbase-mks, 2020-09-16:
Of course, you can.

livemeatz, 2020-09-20:
Problem solved with the new bootloader hex that was recently uploaded. Able to read microSD card, SanDisk 16GB (partitioned @ 4096MB ;default ). Have yet to try wired firmware upload thru ST-LinkV2 or JLink. Thank you very much.

Seelenkind, 2021-02-10:
> Successfully managed to program the stm32 chipset using STLink method (STlinkV2 doggle + software) and flash+verify the firmware using the provided mks_e3_hex file.
Hello, I would like to upload the bootloader for the MKS Robin E3 without a beep. But I can't get into programming mode with J-Flash.
Can you upload the bootloader with the ST-Link dongle without having to change anything on the circuit board?
Did you use the dongle to flash the bootloader or to flash the firmware?
How To Connect to Board ?

[image]

livemeatz, 2021-02-10:
Hello...

Please excuse me.
I used the method from this youtube video

[https://www.youtube.com/watch?v=7O_ZOdnM0SU](url)

and connecting the stlink doggle to the 5 pins of E3 board.

[image]

closed 2021-02-10 by livemeatz

jesterhead82, 2021-04-20:
For Reference to others:

Purchased 3 (!) e3d boards V1.1, all of them did not boot.
None of them accepted firmware update through SD Card.

Purchased cheap ST-LINK V2 Adapter (be careful, some have wrong pinout printed on them), connected it to the board (like livemeatz) using the 5 Pins.

Flashed bootlader from Normal version, flashed own compiled Marlin via ST-Link (Board did not accept any of the provided firmwares on the github via SD card, and board stopped during booting with precomiled Ender3 Version on the Github).
Like mks-wrote you need to set starting adress  0x08005000, i flashed then the firmware.bin from my marlin comiplation (not the robin_e3.bin)

thaers3d, 2021-12-02:
this board have problem , is not accepting firmware .
and the company stop selling it on aliexpress.
they know it have a problem and just they sold there stock to us.
very bad company.