makerbase-mks/MKS-Robin-E3-E3D #33: 3D touch failure

bymanue, 2020-09-05:
Good afternoon
I have a problem with 3d touch, it is connected without problems to the printer.
At the time of doing the test it goes down a first time and it does not fail, but it repeats a second time slower and it fails me randomly at that point. It is connected to the Z limit switch. I have tried the default firm that comes in Github MKS for cr 10 and it also gives me errors. Thank you

makerbase-mks, 2020-09-07:
MKS Robin E3-E3D path Robin_e3.bin file have not config 3D TOUCH function, you need download source code and setting and build, about config 3D TOUCH, you can refer to Wiki :https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH

KiZDGGit, 2020-09-07:
Same here. G29 sometimes finishes fine, but other fails and doesn't stop at the second down, so it's very erratic and completely unusable.

 

> MKS Robin E3-E3D path Robin_e3.bin file have not config 3D TOUCH function, you need download source code and setting and build, about config 3D TOUCH, you can refer to Wiki :https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH

If fails at second time, I suppose it's not "stock" firmware; mine is not.

Any idea?

makerbase-mks, 2020-09-17:
Can you upload config file?

bymanue, 2020-09-17:
KiZD is my neighbour, he is at work now; we've been using the same settings (and same troubles) 

[2. With BLTouch on PB1 and FilamentRunoutSensor.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5241060/2.With.BLTouch.on.PB1.and.FilamentRunoutSensor.zip)

thank you

bymanue, 2020-10-01:
[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5314393/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5314395/Configuration_adv.txt)

makerbase-mks, 2020-10-09:
You can refer to branch: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch

closed 2021-02-22 by mks-viva