makerbase-mks/MKS-Robin-E3-E3D #36: 3d touch and neopixels

almir984, 2020-09-30:
I have successfully installed 3d touch but now I am struggling to make neopixels to work.. I'm using marlin 2.0.6.. on mks robin e3

mustafadh12, 2020-10-01:
can you please share your files, im having trouble on bl touch

makerbase-mks, 2020-10-06:
We will upload it

makerbase-mks, 2020-10-09:
3D TOUCH: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch

LearnIIBurn, 2020-10-15:
Trying to compile with Neopixel active on PA2 pin results in:

`#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"`

That is with MKS Adafruit Neopixel library added in platformio.ini as a `lib_deps` for MKS Robin E3

I could very well have done that wrong but I'm thinking it should have worked ok.

gurues, 2020-11-07:
I have modified marlin as you indicate in the help: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Upgrade-with-Neopixel-lights.
Built only with platformio.ini is OK but built with platformio.ini and configuration.h (neopixel) error.
 Compiling .pio\build\mks_robin_e3\src\src\HAL\STM32F1\SPI.cpp.o
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/SanityCheck.h:52:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"

modified in configuration.h:

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED  // gurues control led
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PA2       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 30       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

Help me please

DocYester, 2021-05-03:
Comment out the mentioned line in SanityCheck.h

Check your platformio.ini for :

```
#
# MKS Robin E3D (STM32F103RCT6) and
# MKS Robin E3 with TMC2209
#
[env:mks_robin_e3]
platform      = ${common_stm32f1.platform}
extends       = common_stm32f1
board         = genericSTM32F103RC
platform_packages = tool-stm32duino
extra_scripts = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/mks_robin_e3.py
build_flags   = ${common_stm32f1.build_flags}
  -DDEBUG_LEVEL=0 -DSS_TIMER=4

```
and add 

```custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel```

Configure the Neopixel as desired in Configuration.h

closed 2021-08-02 by mks-viva