makerbase-mks/MKS-Robin-E3-E3D #38: BL Touch Connection and Update Error

mustafadh12, 2020-10-05:
First,
https://www.youtube.com/watch?v=_x1TzdqGZdo at 02:20 is the connection scheme for BL TOUCH on MKS E3D.
https://www.youtube.com/watch?v=r2A0b7XQEaI&t=1s 04:09 is the OTHER connection scheme for BL Touch on MKS E3D.

These  videos are not useful and make everything complicating. 

Second, 
I've tried both, updated marlin tried everything but nothing worked fine. Now update stucks at 100% . I'm uploading my files for anyone open for helping. 

Third, dear MKS Team, please upload a file for BL TOUCH for your [boards.](url
[Marlin-2.0.x.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5326741/Marlin-2.0.x.zip)

)

Indigo-Tech, 2020-10-07:
same here, just got the board with 3d touch sensor yesterday

I tried compiling the clean version without changes to config files and it still get stuck on 100%, only the bin file that is provided with the repo is working

Indigo-Tech, 2020-10-08:
@mustafadh12 I got by mkr board with 3d touch 2 days ago, installed and encountered the same problem when flushing the firmware the loader stuck at 100%.
so I been looking at the code and found out that I taken the wrong file the output of the compilation is 3 files firmware.bin, firmware.something and the last and the one we need is "Robin_e3.bin", once you take this one this should work as expected,
I recommend not using this repository as it full of mistakes and what not

Here is a fast guide using the official Marlin repo:

- Clone [Marlin repo](https://github.com/MarlinFirmware/Marlin) and [Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations)

- Under Configurations you will find mks e3 for ender 3 example config files take them and replace the ones under Marlin repo

- Use [this video] (https://www.youtube.com/watch?v=7le9L2LMY-U&ab_channel=Crosslink) to configure bltouch

Thats it you ready to go

makerbase-mks, 2020-10-09:
Hi! you can refer to : https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch

mustafadh12, 2020-10-12:
Thank you guys I've finally made it. But here is another problem. When i hit the level bed command, it only probes 1 point and sometimes it goes up instead of going down and probing the point. So that i cant level the bed. Auto home command works but bed leveling still having some bugs. What can i do?

Indigo-Tech, 2020-10-12:
What firmware are you using? Please attach version of the firnware and
configure files

mustafadh12, 2020-10-12:
I've attached the config and the other files. Now when I send level bed, it goes for probing the first point, it goes up instead of downwards to the bed! Than it store the settings in this way. 
[marlin2.0 for Ender3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5363678/marlin2.0.for.Ender3.zip)

makerbase-mks, 2020-10-12:
I want to know, the motherboard wiring is correct?

mustafadh12, 2020-10-12:
It is correct otherwise autohome command shouldn't act downwards. It works properly but level bed command doesnt works fine.

Indigo-Tech, 2020-10-12:
@mustafadh12 I will check later today the diffs between my config and yours .

But as I wrote earlier you better be using official marlin config files, the ones under this repo isn't correct, or confusing.

mustafadh12, 2020-10-12:
I've tried that but it stops and holds at %100 update rate.

Indigo-Tech, 2020-10-13:
@mustafadh12 after you compile what exactly you do? I guess you just take the wrong file

mks-viva, 2020-10-14:
@mustafadh12 If yourself build it and update it, it's work?
About your feedback, we will test it.

mustafadh12, 2020-10-14:
> after you compile what exactly you do? I guess you just take the wrong file, in order to help you I need as much information as possible like what version of platformio you use, what files you copy to where, how you compile the project?

I just put robin_e3.bin file to the sd card, turn on the power supply for the printer and update it as everyone do.

mustafadh12, 2020-10-14:
> @mustafadh12 If yourself build it and update it, it's work?
> About your feedback, we will test it.

As i said it works for the auto home command. Unfortunately, level bed command goes for the first probing point and goes upwards instead of probing.

mks-viva, 2020-10-14:
@

> > @mustafadh12 If yourself build it and update it, it's work?
> > About your feedback, we will test it.
> 
> As i said it works for the auto home command. Unfortunately, level bed command goes for the first probing point and goes upwards instead of probing.

But i test is OK. Can you format the TF card and then update the firmware?

mustafadh12, 2020-10-14:
> @
>
>
> But i test is OK. Can you format the TF card and then update the firmware?

Done it still the same problem unfortunately.

mks-viva, 2020-10-14:
> [image]
> 
> > @
> > > > @mustafadh12 If yourself build it and update it, it's work?
> > > > About your feedback, we will test it.
> > > 
> > > 
> > > As i said it works for the auto home command. Unfortunately, level bed command goes for the first probing point and goes upwards instead of probing.
> > 
> > 
> > But i test is OK. Can you format the TF card and then update the firmware?
> 
> Done it still the same problem unfortunately. I've attached the wiring photo also.

3D TOUCH's Z probe connect to motherboard's E-STOP/PROBE interface, machine's Z-limit connect to Z-STOP like this:
[image]

mustafadh12, 2020-10-14:
First of all it can be wired like this
 https://www.youtube.com/watch?v=_x1TzdqGZdo at 02:20

Now I've wired like the way you said, defined Z_MIN_PROBE_PIN PB1 also, but still the same problem. 

It auto homes, but while leveling bed at the first point it goes upwards and stops.

mustafadh12, 2020-10-20:
any comments? @makerbase-mks @mks-viva

mks-viva, 2020-10-22:
I can't see any difference from the picture. Can you Inital EEPROM: LCD->Configuration->Restore Defaults and after Load Settings

mustafadh12, 2020-10-22:
> I can't see any difference from the picture. Can you Inital EEPROM: LCD->Configuration->Restore Defaults and after Load Settings

I've done it. It probed the first point of other 9 than stopped probbing and said printer is ready.

mks-viva, 2020-10-22:
Does it work properly?

mustafadh12, 2020-10-22:
When I restore settings, it only probes the first point. Than it stores that settings. After that when i command level bed, it starts to go upwards for probing the first point like on the video.

mustafadh12, 2020-11-05:
I'm still waiting for a response please. @makerbase-mks @mks-viva

mks-viva, 2020-11-05:
> I'm still waiting for a response please. @makerbase-mks @mks-viva

We will install 3d touch and test this firmware again, and shoot the leveling video

mks-viva, 2020-12-11:
I has upload 3d touch firmware to here, you can refer to it and you need set borad is BOARD_MKS_ROBIN_E3
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1

closed 2021-04-27 by mustafadh12