makerbase-mks/MKS-Robin-E3-E3D #43: mks robin e3d can not flash and no boot

RutaHunter, 2020-10-29:
Hi. I bought the mks robin e3d board. Downloaded the firmware https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/ender3-3d-touch/firmware/marlin2.0.x%20bin/Robin_e3_TMC2209.bin, renamed Robin_e3.bin, inserted it into the board, turned on the power and no reaction. The blue diode is on and that's it, and there are 2 stripes on the screen. Without a flash drive, the board behaves the same way and does not boot. The bootloader may have broken? How to fix?

mks-viva, 2020-10-30:
Can you provide picture about no boot ?

RutaHunter, 2020-10-30:
[Looks]
[image]
like this all the time. No movement

mks-viva, 2020-10-31:
Hi! The 'boot' will be display only when you use LCD12864 or MKS LCD12864B, so, if you use LCD2004 and it's no display, because bootloader have not LCD2004 display.

RutaHunter, 2020-10-31:
I know lcd2004 doesn't fit. My 12864b broke and until a new one came, I connected lcd2004. I think that in lcd2004 there should also be some movements when executing code in the processor. I know that when the code runs, the red led near the fan connector blink. And in case of successful firmware, the file on the flash card changes the extension to *.cur, but this also does not happen. I have flashed the processor via usart1.

mks-viva, 2020-11-02:
@RutaHunter 
[image]

RutaHunter, 2020-11-03:
Flashing the bootloader didn't help either. Maybe I'm stitching it wrong somehow? Can there be any instructions for the firmware if the firmware does not work with the micro sd?

mks-viva, 2020-11-03:
Can you provide your download bootloader procedure？I will check it.

RutaHunter, 2020-11-03:
Pin boot0 (60) to 3.3v, connect UART to USART1, run STM Flash Loader Demonstrator
[image]
[image]
[image]
[image]
Connect pin boot0 (60) to GND and restart device, but nothing

mks-viva, 2020-11-05:
It seems everything is normal during the burning process

RutaHunter, 2020-11-05:
Any ideas?

mks-viva, 2020-11-05:
I need test it and i will use this flash process and check it work.

mks-viva, 2020-11-05:
I test it and it's work and i use this flash process. So, i think your board 3.3V short to GND, because Boot0 has connect to gnd diretly, you need disconnect it

RutaHunter, 2020-11-05:
Again flash bootloader, disconnect boot 0 from board, insert micro sd with firmware and turned on power, but same

sl1pkn07, 2020-11-05:
> connect UART to USART1

how?

greetings

mks-viva, 2020-11-06:
You need use STM32 PA9-TX1,PA10-RX1(AUX-1 interface)

RutaHunter, 2020-11-06:
> You need use STM32 PA9-TX1,PA10-RX1(AUX-1 interface)
> 
> 1. Disconnect boot0
> 2. USB serial module's TX connect to AUX-1 RX，USB serial module's RX connect to AUX-1 TX，GND connect to GND，5V connect to 5V
> 3. Boot0 connect to 3.3V
> 4. Run STMFlashLoader Demo and add file
> 5. Download
> 
> I has use this flash process and it is work.

I did exactly that, but the program in the processor does not start.

> you mean connect PB6 and PB7 to USB UART dongle?
> 
> https://community.st.com/sfc/servlet.shepherd/version/renditionDownload?rendition=ORIGINAL_Png&versionId=0680X000006qOsl&operationContext=CHATTER&contentId=05T0X00000NPN0r
> 
> is not posible use the SWD port for use with STLink?

[image]

mks-viva, 2020-11-06:
Needn't use STLink, just use a simple USB to serial port module, like this:
[Robin e3 use STMFlashLoader Demo.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5499499/Robin.e3.use.STMFlashLoader.Demo.zip)
[image]
[image]

RutaHunter, 2020-11-06:
I wrote that the processor reads and writes well. It doesn't run the program.

mks-viva, 2020-11-07:
> I wrote that the processor reads and writes well. It doesn't run the program.

When you use boot0 to connect to 3.3V, be sure to disconnect boot0 and GND. If you don’t do this, I have to suspect that the chip is damaged, because your subsequent steps don’t seem to have any problems.

RutaHunter, 2020-11-14:
Hi. I take my 12864lcd/ If i flash to processor this bootloader https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/ender3-3d-touch/firmware/Bootloader/MKS%20Robin%20E3_E3D%20TEST.hex then the screen says "boot" and the speaker beeps.
[image]

if I flash something else, then only the backlight is on on the screen and nothing happens
[image]

mks-viva, 2020-11-17:
You can flash MKS Robin E3_E3D TEST.hex and this hex file include bootloader, and after, you can update Robin_e3.bin file.

RutaHunter, 2020-11-19:
> 
> 
> You can flash MKS Robin E3_E3D TEST.hex and this hex file include bootloader, and after, you can update Robin_e3.bin file.

It did not help

mks-viva, 2020-11-23:
[image]
If this can be started normally, I think it's working ok.

mks-viva, 2020-12-11:
I will close this question temporarily.

closed 2020-12-11 by mks-viva

ostapoff, 2021-03-23:
I have the same issue with my MKS Robin e3d and MKS LCD12864B.

This is how it boots up: https://youtu.be/nOsbkkQBBko

This is how it handles the firmware update: https://youtu.be/mu0gxG2ySPE

Sometimes it is possible to make it start working by inserting an xd card but it works very rarely. And if it works the contrast is too low though it is set to 205. When I get into the corresponding menu the contrast of the screen gets magically repaired: https://youtu.be/z7TEHHgfpps

Also when I attach a screen from the Ender-3 it works like a charm via ext1+ext2 as well as via ext3: https://youtu.be/eUXtF-SC6lw

Of course for the Ender's screen I use **#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER**  and for the MKS LCD12864B the **#define MKS_LCD12864**