makerbase-mks/MKS-Robin-E3-E3D #44: board doesnt response after flashing

2tata, 2020-10-30:
I have a Robin E3 board and wenn I compile https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware/marlin2.0%20for%20Ender3 and continiure flashig the board does not response on serial.

Using Pronterface for testing purpose.

mks-viva, 2020-10-31:
1. You need check COM on PC
2. You need connect motherboard use pronterface by COM and burdrate default is 250000

2tata, 2020-10-31:
> 1. You need check COM on PC
> 2. You need connect motherboard use pronterface by COM and burdrate default is 250000

sorry I was not clearly enough. I already use the stock firmware which works good. But a new compiled firmware did not response. On the tty interface. I also try flashing via wire with JFlash but this software seems not working prober. I tried flashing over wire because the board can only handle SD card in a limited way. Means max size of 4GB or less and fat16 FS. It was not that easy to get an micro SDcard <=4GB.

mks-viva, 2020-10-31:
If you has connect motherboard to PC, it is will display
[image]

2tata, 2020-10-31:
[image]

I just didn't get any response from the board. I think the flashed binary is not working

mks-viva, 2020-10-31:
Can you setting port?
[image]

2tata, 2020-10-31:
It is already the correct device:
```
tata@floki ~ % lsusb          
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 04ca:703c Lite-On Technology Corp. Integrated Camera
Bus 002 Device 049: ID 17ef:100f Lenovo ThinkPad Ultra Dock Hub
Bus 002 Device 048: ID 1532:0109 Razer USA, Ltd Lycosa Keyboard
Bus 002 Device 050: ID 15d9:0a4d Trust International B.V. Optical Mouse
Bus 002 Device 046: ID 17ef:1010 Lenovo ThinkPad Ultra Dock Hub
Bus 002 Device 062: ID 1a86:7523 QinHeng Electronics CH340 serial converter   <---
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
[497828.690033] usb 2-1: USB disconnect, device number 62
[497828.690358] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[497828.690397] ch341 2-1:1.0: device disconnected
[497831.310571] usb 2-1: new full-speed USB device number 63 using xhci_hcd
[497831.452091] usb 2-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[497831.452100] usb 2-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[497831.452103] usb 2-1: Product: USB Serial
[497831.453969] ch341 2-1:1.0: ch341-uart converter detected
[497831.454638] usb 2-1: ch341-uart converter now attached to ttyUSB0
[497831.930000] TCP: request_sock_TCP: Possible SYN flooding on port 34349. Sending cookies.  Check SNMP counters.
```

mks-viva, 2020-10-31:
Bus 002 Device 062: ID 1a86:7523 QinHeng Electronics CH340 serial converter   <---

mks-viva, 2020-10-31:
Have you used a Raspberry Pi?

2tata, 2020-10-31:
> Have you used a Raspberry Pi?

no I haven't. In which way should I use a rpi?

I am working on my desktop system.

mks-viva, 2020-11-02:
> > Have you used a Raspberry Pi?
> 
> no I haven't. In which way should I use a rpi?
> 
> I am working on my desktop system.

MKS Robin E3 motherboard's USB driver is CH340, so, you can check it your COM on PC

2tata, 2020-11-02:
> > > Have you used a Raspberry Pi?
> > 
> > 
> > no I haven't. In which way should I use a rpi?
> > I am working on my desktop system.
> 
> MKS Robin E3 motherboard's USB driver is CH340, so, you can check it your COM on PC

The serial is not the problem. I already proof this by using the stock firmware. the problem seems to be in the compiled binary.
[firmware.bin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5474339/firmware.bin.zip)

This binary was compiled out of tho about mentenorid repo.

mks-viva, 2020-11-03:
Oh! You need copy Robin_e3.bin file to TF card and insert motherboard and update it. Is not copy firmware.bin file
[image]

2tata, 2020-11-03:
> Oh! You need copy Robin_e3.bin file to TF card and insert motherboard and update it. Is not copy firmware.bin file
> [image]

Yes, I exactly did how it was described, and I got an CRC file back but still the compiled binary, means the outcoming "firmware.bin" which was renamed to "Robin_e3.bin" seems not working.

mks-viva, 2020-11-03:
You cannot directly use firmware.bin to rename Robin_e3.bin, because these two files are not exactly the same. Robin_e3.bin is transcoded on the basis of firmware.bin

2tata, 2020-11-03:
> You cannot directly use firmware.bin to rename Robin_e3.bin, because these two files are not exactly the same. Robin_e3.bin is transcoded on the basis of firmware.bin

ah, I didnt saw any hint to the transcoding :). How I can do that?

mks-viva, 2020-11-03:
When you compile the firmware, marlin has already done this work, so you only need the cpoy Robin_e3.bin file to update the firmware

2tata, 2020-11-03:
> When you compile the firmware, marlin has already done this work, so you only need the cpoy Robin_e3.bin file to update the firmware

yes, I saw it right before your message. Classical Layer 8 problem ;-P sorry for the noise.

closed 2020-11-03 by 2tata