makerbase-mks/MKS-Robin-E3-E3D #45: ERROR CONFIGURATION NEOPIXEL

gurues, 2020-11-07:
I have modified marlin as you indicate in the help: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Upgrade-with-Neopixel-lights.
Built only with platformio.ini is OK but built with platformio.ini and configuration.h (neopixel) error.
Compiling .pio\build\mks_robin_e3\src\src\HAL\STM32F1\SPI.cpp.o
In file included from Marlin\src\HAL\STM32F1../../inc/MarlinConfig.h:50:0,
from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1../../inc/../HAL/STM32F1/inc/SanityCheck.h:52:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"

modified in configuration.h:

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED // gurues control led
#if ENABLED(NEOPIXEL_LED)
#define NEOPIXEL_TYPE NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
#define NEOPIXEL_PIN PA2 // LED driving pin
//#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
//#define NEOPIXEL2_PIN 5
#define NEOPIXEL_PIXELS 30 // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
#define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
#define NEOPIXEL_BRIGHTNESS 127 // Initial brightness (0-255)
//#define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup

// Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
//#define NEOPIXEL2_SEPARATE
#if ENABLED(NEOPIXEL2_SEPARATE)
#define NEOPIXEL2_PIXELS 15 // Number of LEDs in the second strip
#define NEOPIXEL2_BRIGHTNESS 127 // Initial brightness (0-255)
#define NEOPIXEL2_STARTUP_TEST // Cycle through colors at startup
#else
//#define NEOPIXEL2_INSERIES // Default behavior is NeoPixel 2 in parallel
#endif

Help me please

gurues, 2020-11-08:
I have commented on this line en src/HAL/STM32F1/inc/SanityCheck.h

`#if ENABLED(NEOPIXEL_LED)
  //#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
#endif
`
The build is correct but i am afraid of damaging the board.  Could you tell me if it is correct and I can load the firmware

Library Adafruits Neopixel download of github mks but  i have this advertences:
`
Linking .pio\build\mks_robin_e3\firmware.elf
.pio/build/mks_robin_e3/src/src/feature/leds/leds.cpp.o: In function `LEDLights::setup()':
D:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\feature\leds/leds.cpp:68: warning: undefined reference to `Marlin_NeoPixel::init()'
.pio/build/mks_robin_e3/src/src/feature/leds/leds.cpp.o: In function `LEDLights::set_color(LEDColor const&, bool)':
D:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\feature\leds/leds.cpp:102: warning: undefined reference to `Marlin_NeoPixel::set_color(unsigned long)'
Building .pio\build\mks_robin_e3\firmware.bin
.pio/build/mks_robin_e3/src/src/feature/leds/leds.cpp.o: In function `LEDLights::set_color(LEDColor const&, bool)':
Checking size .pio\build\mks_robin_e3\firmware.elf
D:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\feature\leds/leds.h:95: warning: undefined reference to `Marlin_NeoPixel::adaneo1'
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
.pio/build/mks_robin_e3/src/src/feature/leds/printer_event_leds.cpp.o: In function `pel_set_rgb(unsigned char, unsigned char, unsigned char)':
RAM:   [===       ]  27.6% (used 13560 bytes from 49152 bytes)
D:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\feature\leds/printer_event_leds.cpp:54: warning: undefined reference to `Marlin_NeoPixel::adaneo1'
Flash: [========= ]  88.0% (used 23081.pio/build/mks_robin_e3/src/src/gcode/feature/leds/M150.cpp.o: In function `GcodeSuite::M150()':
2 bytD:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\gcode\feature\leds/M150.cpp:69: warning: undefined reference to `Marlin_NeoPixel::adaneo1'
es from 262144 bytes)
D:\Oscar\Impresion 3D\Marlin\Marlin 2.0.7.1\Marlin 2.0.7.1 MKS Robin-V6-BMG-BLTouch-Neopixel/Marlin\src\gcode\feature\leds/M150.cpp:69: warning: undefined reference to `Marlin_NeoPixel::neoindex'
encrypt([".pio\build\mks_robin_e3\firmware.bin"], [".pio\build\mks_robin_e3\firmware.elf"])
================================================= [SUCCESS] Took 260.61 seconds =================================================
Environment    Status    Duration
-------------  --------  ------------
mks_robin_e3   SUCCESS   00:04:20.614
`
Conexión - > [https://drive.google.com/file/d/15vXW9niBGBLkNVABRt4TjmetIYwE4iNc/view?usp=sharing](url) 

Thanks

DocYester, 2021-05-03:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/36#issuecomment-831067007

mks-viva, 2021-06-16:
You need add lib link in Robin e3 ini file
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
[image]

mks-viva, 2021-08-03:
The latest version marlin, you can directly define NEOPIXEL_LED and pin

closed 2021-08-03 by mks-viva