makerbase-mks/MKS-Robin-E3-E3D #46: Hello please help me restore factory settings

lcz0210, 2020-11-08:
Firmware flashing, I made a mistake, I want to go back to the motherboard when I bought it

mks_robin_e3. My machine is ender-3 pro

Can you tell me where to download the firmware

mks-viva, 2020-11-09:
You can download it form: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/marlin2.0.6%20bin

lcz0210, 2020-11-09:
老板 你给我发的这个BIN文件 没办法修改里面的参数啊  我需要安装你们家购买的 3Dtouch 和自动关机模块  修改参数才可以使用的
可以给我发一下  可以修改固件的参数的 那个固件吗

mks-viva, 2020-11-09:
我压缩一个放这里，直接从github下载还是有点慢

lcz0210, 2020-11-09:
非常感谢  困扰我6天的烦恼终于解决了 都想放弃这个板子了

mks-viva, 2020-11-09:
你可以问我们销售人员，也可以直接在这边提问题，下次早点

closed 2020-11-09 by mks-viva

Ferropluss, 2025-01-17:
Здравствуйте. А есть заводские настройки для QIDI X-MAX3? 

[image] - моя материнская плата.