makerbase-mks/MKS-Robin-E3-E3D #50: Is there support to mount the SDcard on the Robin E3 through USB?

hieppo, 2020-12-05:
I connect the Robin E3 to a Pi and been trying to mount the SDcard on the E3. However, I do not see any device listed. Is the Marlin firmware has support for this feature? Is there a Smoothieware for this board?

mks-viva, 2020-12-05:
Hi! I am very sorry, the current MKS Robin E3 motherboard does not support this function, mainly because the Robin E3 motherboard uses the serial port chip instead of directly using the STM32 virtual serial port function.
[image]

hieppo, 2020-12-05:
Hmm, disappointed but thanks for the information. I guess I have to pop the sdcard out  each to flash. Is there a flash method using usb?

mks-viva, 2020-12-07:
USB is currently only used to connect to the host computer and cannot perform other functions.

mks-viva, 2020-12-11:
We will improve it in the next version, so stay tuned. I will close this issues

closed 2020-12-11 by mks-viva