makerbase-mks/MKS-Robin-E3-E3D #51: Fillament cooling fan

edvard5, 2020-12-07:
When connected fillament cooling fan BL touch stopped working, when initializing probe, fan LED turns on for half a second and turns of, the probe not extending, auto level not working. Flashed latest software to TFT35 and motherboard itself. Second time this happens, might be due to fan pinout not having and diode to protect from voltage coming back to motherboard as my setup has stepdown voltage module which is working fine. Second board fried I guess... Might be firmware related issue. Maybe someone have the same issue as me. Thanks in advance.

mks-viva, 2020-12-08:
Hi! FAN runs for half a second, this is normal, this is marlin settings
About BLTOUCH or 3D TOUCH, you can refer to firmware: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1
If you use MKS Robin E3 V1.0 vserion, you need define motherboard is BOARD_MKS_ROBIN_E3 in configuration.h file

mks-viva, 2020-12-11:
If there is no response, then I will temporarily close this issues

closed 2020-12-11 by mks-viva