makerbase-mks/MKS-Robin-E3-E3D #52: Frimeware creality ender 5

Fogs135, 2020-12-10:
Bought e3d board for Ender 5. But no firmware for this printer model.  can you give the firmware for ender 5.

Fogs135, 2020-12-13:
The problem is closed.  Adapted firmware from skr 1.3 for tmc2209 and bltouch.

closed 2020-12-13 by Fogs135

mks-viva, 2020-12-14:
Can you share the Robin E3D ender5 machine firmware, I will put it on Github, and let more people use it. I will be very grateful.

Fogs135, 2020-12-14:
well within 2-3 working days will send you.

mks-viva, 2020-12-14:
Ok, looking forward to your good news, thank you!

Fogs135, 2020-12-15:
[Marlin-2.0.x.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5697066/Marlin-2.0.x.zip)
I changed my motherboard type. Adapted all axes movements for Ender 5.
The firmware is configured for the TMC 2209 v2.0 driver in the UART.
Bltouch activated with Bilinear.
Russian language.
After installing the firmware, you need to calibrate the position of the Bltouch sensor yourself, as well as the number of steps per mm for the Z axis and Extruder through the printer menu.

azatalp, 2021-02-05:
How can I upgrade my mother board with your files? (ender 5 mks e3)

mks-viva, 2021-02-06:
@azatalp You can download Fogs135's firmware