makerbase-mks/MKS-Robin-E3-E3D #53: display tft 2.4 commands does not work via display

diegomsdos, 2020-12-13:
I have an ender 5 printer and I changed the card for robin e3d mks. I have a problem with the fimeware that comes on the card because it is from ender 3. I updated the firmeware for marlin 2.0.7.2 which is the most recent version after that. commands via tft 2.4 no longer works.

note: I used the files config.h and config.adv.h that comes in the marlin examples folder this file was for address 3 I made the inversion of the X and Y axes to be able to make the address 5 is in the correct position

I would like to know which command I have to activate to operate the commands via touch

mks-viva, 2020-12-14:
Can you provide your configuration.h file to me?
1、I think you need set port
#define SERIAL_PORT 1   //Because MKS Robin E3/E3D use STM32 usart1
2、And keep the baud rate the same
#define BAUDRATE 250000

diegomsdos, 2020-12-14:
Thanks, I set BAUDRATE to 250000 and it worked

mks-viva, 2020-12-15:
This is good news. So I will close this issues.

closed 2020-12-15 by mks-viva