makerbase-mks/MKS-Robin-E3-E3D #54: Robin E3 with BL touch stuck on Marlin boot logo

bulcke, 2020-12-13:
Trying to get BL touch working. 

So I tried modifying https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/V1.1/Marlin-2.0.6.1_for_ender3_TMC2209%2B3dtouch.rar.
Only changed the offsets to: #define NOZZLE_TO_PROBE_OFFSET { -48.3, -7, -2.9}

This compiles ok.

Upgrading with ".pio\build\mks_robin_e3\Robin_e3.bin" works fine. But then display hangs on the Marlin boot logo forever. Reboot does not help.

This firmware https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/V1.1/Firmware%20bin%20for%20ender3/Marlin-2.0.6.1_for_ender3_TMC2209%2B3dtouch.bin boots fine. BL touch is working with this one. But stock offsets (#define NOZZLE_TO_PROBE_OFFSET { -44, -14.4, -2.4}) are not of much use for me.

mks-viva, 2020-12-14:
Hi! Can you provide Video or picture to me?

bulcke, 2020-12-14:
> 
> 
> Hi! Can you provide Video or picture to me?

I'll try next weekend.

mks-viva, 2020-12-15:
Okay, but I still need to confirm:
Will this problem occur if you use the MKS github Robin_e3.bin file?
If not, then it will be due to software settings

bulcke, 2020-12-21:
Sorry, found no time to look into this yet...

mks-viva, 2020-12-22:
When you have time, you can test it again

RJD2two, 2021-01-03:
I tried Marlin-2.0.6.1_for_ender3_TMC2209+3dtouch.bin and it hangs at the boot screen for me too. Just displays marlin 2.0.6.1

RJD2two, 2021-01-03:
hoiwever, this one does work with the probe! 
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/ender3-3d-touch/firmware/marlin2.0.x%20bin/Robin_e3_TMC2209_3DTOUCH_PROBE.bin

bulcke, 2021-01-09:
Trying again, from scratch, with https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/V1.1/Marlin-2.0.6.1_for_ender3_TMC2209%2B3dtouch.rar

I'm using visual studio code for editing and compiling.

Doublechecked the firmware settings according to this https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH

Just a heads up for any future readers: In that guide it says:
`
9.Set distance between probe and the edge of the hotbed to prevent the probe from exceeding the hotbed

    Configuration.h：#define **MIN_PROBE_EDGE** 10

10.Enable leveling range

    Configuration_adv.h

#if PROBE_SELECTED && !IS_KINEMATIC
  #define MIN_PROBE_EDGE_LEFT **MIN_PROBE_EDGE**
  #define MIN_PROBE_EDGE_RIGHT **MIN_PROBE_EDGE**
  #define MIN_PROBE_EDGE_FRONT **MIN_PROBE_EDGE**
  #define MIN_PROBE_EDGE_BACK **MIN_PROBE_EDGE**
#endif

`
But in the current code **MIN_PROBE_EDGE** is defined as **PROBING_MARGIN**. It is referenced correctly in Configuration_adv.h however, so no change needed.

This time I did not change the line "#define NOZZLE_TO_PROBE_OFFSET { -44, -14.4, -2.4}" in configuration.h .

The only thing I did change in Configuration.h is "#define GRID_MAX_POINTS_X 4"

Compiling was succesful.
Terminal only threw a few warnings, have seen these before, should not be the issue here.

`C:\users\didi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;
                              ^~~~~~~~Compiling .pio\build\mks_robin_e3\lib3c4\USBComposite\Joystick.cpp.o      
~~

...

C:\users\didi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c: 
In function 'xPortStartScheduler':
C:\users\didi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:178:2: warning: implicit declaration of function 'systick_attach_callback' [-Wimplicit-function-declaration]     
  systick_attach_callback(&xPortSysTickHandler);
  ^~~~~~~~~~~~~~~~~~~~~~~
At top level:
C:\users\didi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:285:6: warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )
      ^~~~~~~~~~~~~~~~~~~~~~
`

Video of flashing with .pio\build\mks_robin_e3\Robin_e3.bin : https://youtu.be/DcMWxpWhK6U

Not much to see actually, you see the update % going up and the device rebooting.
You'll notice on the boot logo it does show the new version number 2.6.0.1.
I stopped the video but left the printer like that another 10 minutes. No change. Power cycling (without the SD card) does not help.

bulcke, 2021-01-09:
> 
> 
> hoiwever, this one does work with the probe!
> https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/ender3-3d-touch/firmware/marlin2.0.x%20bin/Robin_e3_TMC2209_3DTOUCH_PROBE.bin

That one works for me too. But the issue still remains if I can't flash the one I alter and compile myself.

mks-viva, 2021-01-11:
@bulcke 
Hi! According to your test feedback, it is largely caused by the compilation environment. Or you modify the parameters, resulting in too much program space. Especially for grid points, you can compile first according to the default of MKS firmware.

bulcke, 2021-01-11:
I'm not sure what you mean with program space, but I find it hard to believe that a single small change would cause this. Ok so a 4x4 grid needs to hold 7 more probing points than a 3x3... big deal.

But I'll try anyway, without changing anything in the configs. Will get back to you as soon as I find some time.

mks-viva, 2021-01-12:
Ok, waiting for your good news.

closed 2021-08-03 by mks-viva