makerbase-mks/MKS-Robin-E3-E3D #55: TMC Connection Error

ricardonato, 2020-12-16:
I have connected the TMC2209 v1.0 drivers to my new MKS Robin E3D (003) board without modifying anything.

I have updated the "firmware bin for Ender3 (TMC2209)"
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware/firmware%20bin%20for%20Ender3(TMC2209)

When Marlin finishes booting, the message appears:
"TMC Connection Error"

mks-viva, 2020-12-17:
Hi! You need to do the following so that TMC will not report errors
1. Usrt mode set in E3D motherboard（Use jumpers connect it）
[image]
2. You need use DC12-24V power

ricardonato, 2020-12-17:
Thank you. Problem solved

closed 2020-12-17 by ricardonato