makerbase-mks/MKS-Robin-E3-E3D #56: error compilation

ptitpuma, 2020-12-24:
hi i have a trouble with compiling marlin with vs code for a cr10
i've follow every step but it won't compile due to this error :
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2378:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
     #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:

if i change value to1 it works...

mks-viva, 2020-12-25:
Hi! I know what setting causes, you need set
#define SERIAL_PORT 1  //For USB
Disable #define SERIAL_PORT_2 1

ptitpuma, 2020-12-27:
ok thanks

mks-viva, 2020-12-28:
Can it be compiled normally?

ptitpuma, 2020-12-28:
Yes with serial port 1...
don't know why they say to write -1

mks-viva, 2020-12-29:
The first version of marlin is defined as -1, only the latest version will define 1, otherwise it will not compile, Marlin firmware has been adjusted. So, i will close this issues temporarily

closed 2020-12-29 by mks-viva

ptitpuma, 2021-01-31:
hi just another question about that
hw should i configurate this to add a tft 35 ?

Niptoc, 2021-03-30:
Hello

I also have compilation error for a CR10.
I want to use the TFT 35 and I got the error message :
`In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:839:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."`

I have activated the ADVANCED_PAUSE_FEATURE to be able to use the filament end detector.

Do anyone encounter the same issue?