makerbase-mks/MKS-Robin-E3-E3D #58: MKS ROBIN E3D z axis not respond after update the firmware

Xuculate, 2020-12-28:
Hi I'm a little desperate:

Hi. I have recently installed the MKS Robin E3D board +2209.

The board worked fine with the Marlin 2.0.6.1 stock firmware. But I wanted to reverse the rotation of the extruder motor because I have installed a BMG. I have tried to put the firmware from here: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware and now the z axis does not work, also I still do not know how to compile the Marlin configuration for reverse the extruder motor, etc. I am totally new to this Marlin thing. If you have a configuration for the CR10S to work fine it would be greatly appreciated.

I have changed the controllers between them and they work fine, I have also changed the cables and the output of z does not give signal. So I don't know if it's because of firmware or hardware.

Another things that I´ve tried:
-I've changed the cable outputs from Z cables on other motors, x and y and don´t respond
-I have connected the Z motors on the X and Y outputs of the board and if they move, even using the Z cables, and the Z motors work, so it's not the fault of either the 2209 drivers, the cables, the motors, or it's a firmware issue or the board has been broken inexplicably.

I don't know what to do now I have the printer inoperative, and the Creality board I think has also been broken since before putting the MKS I tried to reverse the extruder cable and never worked the extruder output again. It happens as is happening with the MKS board that the extruder port does not give signal, but in this case the MKS cables I have always connected them in their right position.

I don't know what to do. For the moment I am waiting for a new board (E3) to try again.

Thank you!
[image]

mks-viva, 2020-12-29:
Are you using MKS Robin E3D V1.1 version motherboard? If yes, then the Z drive signal has been modified and you should download this firmware
Link：https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1
[image]

MKS Robin E3D/E3 V1.1 and V1.0 compare
https://github.com/MarlinFirmware/Marlin/pull/20216
[image]

Xuculate, 2020-12-29:
WOW! Yes it is solved! thank you very much!

Now I need invert the direction of the extruder motor. What of this is it?

// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false
#define INVERT_E5_DIR false
#define INVERT_E6_DIR false
#define INVERT_E7_DIR false

mks-viva, 2020-12-30:
Haha! So, i will close this issues temporarily.

closed 2020-12-30 by mks-viva

kjanek, 2021-11-06:
> Haha! So, i will close this issues temporarily.

Please change this too in klipper config file. It took me literally one whole day to find this issue and resolve not moving z axis after changing from marlin to klipper on the mks robin e3d v1.1