makerbase-mks/MKS-Robin-E3-E3D #59: Firmware Upgrade not possible E3d V1.0

Seelenkind, 2020-12-28:
Hi there ,
I cannot upload firmware to the MKS board.
When I save the firmware on an SD card and switch on the printer, it counts up from 0% to 100% and then nothing happens.
When I pull out the SD card and restart the printer, the display only shows boot. But nothing happens.
Do you have a solution? Is the board defective?

mks-viva, 2020-12-29:
Hi! 
1. If TMC Connect Error, you need use 12/24V power on
2. Update firmware, you need acooding to your screen and download firmware bin file, you can refer to here
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.0/firmware%20bin

Seelenkind, 2020-12-29:
> Hi! 
> 1. If TMC Connect Error, you need use 12/24V power on
> 2. Update firmware, you need acooding to your screen and download firmware bin file, you can refer to here
> https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.0/firmware%20bin
> [image]
> 

I got the firmware exactly from this site.  after the 100% I see on the SD card that the file name has been changed to.CUR the problem is not that I can not find the firmware but that it does not go on after the 100%.  after a restart the display only shows boot.  but does not go any further.

mks-viva, 2020-12-30:
Can you provide your test firmware bin to me? And i will test it.

Seelenkind, 2021-01-01:
> Can you provide your test firmware bin to me? And i will test it.

Hi there,
I think it's because the firmware file that I downloaded is only 103 KB in size. A normal fiemware has over 200 KB. Can it be that the finished firmware on the Github page is faulty?

Seelenkind, 2021-01-01:
> Can you provide your test firmware bin to me? And i will test it.

OK problem solved. If I use the right mouse button to download the firmware file, the file is only 104 KB in size. If I click on the firmware file and use the download button, it is 205 KB in size and it can be uploaded.
The problem is solved and the question can be marked as solved.

mks-viva, 2021-01-03:
Ok! I will close this issues temporarily.

closed 2021-01-03 by mks-viva