makerbase-mks/MKS-Robin-E3-E3D #60: MKS ROBIN E3D FILAMENT SENSOR FIRMWARE configuration

Xuculate, 2020-12-29:
Hello I am tring to activate the filament sensor in the Marlin firm but I have problems to compile. I was trying to follow this tutorial:

https://www.youtube.com/watch?v=6Hk5Zk_G968&ab_channel=MakerbaseTeam

This was my error:

[image]

mks-viva, 2020-12-30:
Hi! Can you provide your firmware to me? And i need test it.

Xuculate, 2020-12-30:
> Hi! Can you provide your firmware to me? And i need test it.

Yes sure

Here is the file:
https://gitlab.com/Xuculate/cr10s-mks-robin-e3d-v1.1_bmg_sensor-filament/-/blob/master/Marlin-2.0.6.1_for_ender3_TMC2209.rar

henryho8, 2021-01-17:
I got this problem too. Since I updated my Platform IO with new version, I downloaded E3D marlin firm and failed to compiling,
[image]

henryho8, 2021-01-17:
> I got this problem too. Since I updated my Platform IO with new version, I downloaded E3D marlin firm and failed to compiling,
> [image]

I tried to disable filament runout and failed again with this message.
[image]

henryho8, 2021-01-17:
> > I got this problem too. Since I updated my Platform IO with new version, I downloaded E3D marlin firm and failed to compiling,
> > [image]
> 
> I tried to disable filament runout and failed again with this message.
> [image]

I don't know what i did...
[image]
;