makerbase-mks/MKS-Robin-E3-E3D #61: Incorrect serial port -1

ellensp, 2021-01-01:
The Example configuration file MKS-Robin-E3-E3D/firmware/marlin2.0 for Ender3/Marlin/Configuration.h
 has #define SERIAL_PORT_2 -1

This is Incorrect

-1 can only be used when the USB D+ and D- lines are directly connected to the Controllers MPU pins.

This board uses a usb/seral converter chip.

mks-viva, 2021-01-03:
Hi! Thank you for your feedback, we will test it and fix it.

closed 2021-02-01 by mks-viva

ellensp, 2021-02-01:
Ignoring the issue and sticking your head in the sand is not fixing it!

mks-viva, 2021-02-01:
> Ignoring the issue and sticking your head in the sand is not fixing it!

https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/61#issuecomment-770527700
I has disable it

mks-viva, 2021-02-01:
At that time, Test firmware can be used normally, and the device corresponding to the value of -1 is not actually considered

ellensp, 2021-02-01:
With more modern marlin it knows the controller doesn't have USB CDC serial and errors to tell you not to set -1. Users where copying these older configs into modern marlin and it would fail (as it should) they then create issue that marlin was broken because you had -1 in your config so it must be correct  (sigh) 

All good now, thanks