makerbase-mks/MKS-Robin-E3-E3D #62: How to upload bootloader ?

Seelenkind, 2021-01-01:
Hi, 
how to upload bootloader ? 
i want to slience the beep at start

https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware/Bootloader

i connected PC with MKS Board with FTD232 USB to Serial Modul. 
but Flash Loader Demonstrator can not find Board , but connected as Com Port 8

thx

mks-viva, 2021-01-03:
Hi! https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/43

Seelenkind, 2021-01-03:
> Hi! #43
Hi,
I already found this question in the archive and am not sure which of these pins should be pin 60?
Pin 60 should either be on 3.3. set or the connection to GND disconnected. But which one is it exactly?
Why isn't a jumper installed on the board to put the board in bootloader upload mode?
[image]

mks-viva, 2021-01-04:
Hi! About 60pin map
[image]
This may be something we did not consider, I am sorry. But MKS Robin E3/E3D has TF update firmware function

mks-viva, 2021-01-18:
Hi! If there is no response, then I will temporarily close this question.

closed 2021-01-18 by mks-viva