makerbase-mks/MKS-Robin-E3-E3D #64: Issue with Ender3 TMC2209+3DTouch

Benji198599, 2021-01-12:
After I have installed the firmware "For Ender3 TMC2209 + 3Dtouch" via an SD card and want to approach the reference via "AutoHome", the printer moves to the stop of the X axis. The stop for reference X no longer works.

3D Touch was connected as described here: https://www.youtube.com/watch?v=r2A0b7XQEaI&t=177s

The firmware was not changed and only from: https://github.com/makerbase-mks/MKS-Robin-E3-E3D
downloaded.
[image]

Hardware: Ender 3 pro, with MKS Robin E3 V1.0 Motherboard. 
Standard LCD Display

https://user-images.githubusercontent.com/76880195/104358275-38878d00-5534-11eb-853c-4e7b15d65a2d.mov

mks-viva, 2021-01-14:
Can you view the error message on the display screen or other?

Benji198599, 2021-01-20:
There is no error on display... I upload again the firmware https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware/marlin2.0.x%20bin
Robin_e3_TMC2209_3DTOUCH_PROBE.bin

now Z-axis is working but 3DTouch is off and don´t work anymore
[image]
[image]

[image]
[image]

Benji198599, 2021-01-20:
[image]

Benji198599, 2021-01-20:
3DTouch have no light on - no action if I use 3DTouch option in Configuration

mks-viva, 2021-01-25:
Your 3D TOUCH's Z-out connect to Probe/E-STOP(PB1) interface
[image]

krzysziekz, 2021-02-26:
Double check your connections, not wires but sockets, I think they are wrong. You connected probe to neopixel socket
[image]

zeck7, 2021-09-17:
> After I have installed the firmware "For Ender3 TMC2209 + 3Dtouch" via an SD card and want to approach the reference via "AutoHome", the printer moves to the stop of the X axis. The stop for reference X no longer works.
> 
> 3D Touch was connected as described here: https://www.youtube.com/watch?v=r2A0b7XQEaI&t=177s
> 
> The firmware was not changed and only from: https://github.com/makerbase-mks/MKS-Robin-E3-E3D
> downloaded.
> [image]
> 
> Hardware: Ender 3 pro, with MKS Robin E3 V1.0 Motherboard.
> Standard LCD Display
> 
>  IMG-9053.mov

Hi have you solve this issue. I think i got the same problems that X Axis always moving in one direction which is to the endstop. Usually if i choose home...all 3 axis should be move to the endstop until triggered the microswitch and move a bit to OPPOSITE direction to avoid collision with endstop and move slowly back to the endstop until all axis triggered the switch.

But my X Axis acting weird because if im homing all Axis the X Axis only moving to the endstop, and after its triggered the limit switch....it move more to the endstop and grinding with the limit switch. Should it move to the opposite to avoid grinding the axis like in the video?