makerbase-mks/MKS-Robin-E3-E3D #66: Home and move issues

edvard5, 2021-01-17:
After Homing axis, move doesn't work, 3D touch is not leveling. End stops working fine, tested.

mks-viva, 2021-01-18:
Hi! This looks like the program has stopped running after returning to home. You can try to use USB to connect to the motherboard, and the host computer can check the printed information after returning to home, and you can locate what caused the axis to be unable to move.

edvard5, 2021-01-18:
I'll give it a try and report back

mks-viva, 2021-01-18:
Ok!

edvard5, 2021-01-18:
[ERROR] Error:!! STOP called because of BLTouch error - restart with M999
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
two errors

mks-viva, 2021-01-18:
Ok! Please tell me which firmware are you using?

edvard5, 2021-01-18:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/V1.1/Firmware%20bin%20for%20ender3/Marlin-2.0.6.1_for_ender3_TMC2209%2B3dtouch%2BZ_SAFE_HOMING.bin
this one with TFT35 1.0.7v

mks-viva, 2021-01-18:
3D Touch's output signel need connect to probe/E-stop interface
[image]

edvard5, 2021-01-18:
[image]
Already connected, followed step by step tutorial

edvard5, 2021-01-18:
Probelm still persists. need help 🌵

mks-viva, 2021-01-18:
I am sorry! I seem to be doing the wrong guide.
[image]

mks-viva, 2021-01-18:
You need remove ender3 original Zstop

edvard5, 2021-01-18:
offset set corectlly but it keeps pushing nozzle to bed with force until starts to skip

edvard5, 2021-01-18:
after homing still same problem, can't manually move axis... like in the video

edvard5, 2021-01-20:
managed to fix this with these steps
M502
M500
M999
then setting xyz offsets all over again.

closed 2021-01-20 by edvard5

mks-viva, 2021-01-20:
I didn't think about it,