makerbase-mks/MKS-Robin-E3-E3D #67: Bed size values incorrect

ricardonato, 2021-01-23:
I've been going crazy for two weeks. I didn't know why my Ender3 with MKS Robin E3D was not printing the parts to the right and the back of the bed well.

I have seen in the source code of your firmware that in Marlin's "Configuration.h" the values are incorrect:
# define X_BED_SIZE 220
# define Y_BED_SIZE 220

The correct values are:
# define **X_BED_SIZE 235**
# define **Y_BED_SIZE 235**

I have not been able to change them because when I compile with VSCode it returns a lot of errors and I am still working on it.

Can you update those values so that I can download a firmware binary file?

Thank you

mks-viva, 2021-01-25:
Hi! You need to tell me which firmware are you using now?

ricardonato, 2021-01-25:
> ¡Hola! ¿Necesitas decirme qué firmware estás usando ahora?

https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/ender3-3d-touch/firmware/firmware%20bin%20for%20Ender3(TMC2209)/Robin_e3.bin

mks-viva, 2021-01-26:
[Robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5872825/Robin_e3.zip)
I has change it and build it.
[image]

ricardonato, 2021-01-26:
Thank you very much for the support.

closed 2021-01-26 by ricardonato