makerbase-mks/MKS-Robin-E3-E3D #69: E3D "power supply" and "12-24V output" connectors

fantomsbg, 2021-01-30:
My original Ender 3 board is v4.2.2
[image]
Its power connector is located to the E3D "12-24V output" place. So I have to connect the power leads to it and the fan leads to "power supply" connector.
Is it safe and will the fuses fork correctly if the board is power supplied from the "12-24V output" connector?

mks-viva, 2021-02-01:
Hi! 12/24 output interface depends on the power supply voltage you provide.

closed 2021-02-01 by mks-viva

fantomsbg, 2021-02-01:
My question was if I can connect the power supply leads to 12-24V output connector?

mks-viva, 2021-02-02:
Yes, you can.

fantomsbg, 2021-02-02:
Is there a fuse to protect this "12-24V output" when it become an input power supply connector?
If there are no fuses this can cause a serious problems?
According to me the fuse is only for the "power supply" connector.