makerbase-mks/MKS-Robin-E3-E3D #70: Trouble configuring ONBOARD function to use the SD Card directly without a LCD

HarpoDog, 2021-01-30:
I am trying to configure Marlin 2.0 Bugfix downloaded 1.30.21 to use the autostart feature directly from the SD card reader. 
I completed the configuration for a MKS Robin E3 with the Creality Ender 3 configuration files and it complied fine.

Next, I went to the configuration_adv.h file and uncommented  #define SDCARD_CONNECTION ONBOARD on line 1403 and tried to build Marlin and got the following error message three places:
Marlin\src\HAL\STM32F1\onboard_sd.cpp:42:25: error: 'ONBOARD_SD_CS_PIN' was not declared in this scope

Could you please tell provide me the proper files that I could copy into Marlin to remove this error or tell me which files I would need to modify, where within the files they would need to be modified, and the exact code I would need to use.

I am ok if the LCD gets disable as I will not have it attached.

Attached below are my configuration.h and configuratin_adv.h files.

Here is the complete error message:
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:16:
Marlin\src\HAL\STM32F1\onboard_sd.cpp: In function 'void deselect()':
Marlin\src\HAL\STM32F1\onboard_sd.cpp:42:25: error: 'ONBOARD_SD_CS_PIN' was not declared in this scope
 #define CS_HIGH() WRITE(ONBOARD_SD_CS_PIN, HIGH) /* Set OnboardSPI cs high */
                         ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = _BV32(PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:134:3: note: in expansion of macro 'CS_HIGH'
   CS_HIGH();    /* CS = H */
   ^~~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:42:25: note: suggested alternative: 'ONBOARD_SD_SPI'
 #define CS_HIGH() WRITE(ONBOARD_SD_CS_PIN, HIGH) /* Set OnboardSPI cs high */
                         ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = _BV32(PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:134:3: note: in expansion of macro 'CS_HIGH'
   CS_HIGH();    /* CS = H */
   ^~~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp: In function 'int select()':
Marlin\src\HAL\STM32F1\onboard_sd.cpp:41:25: error: 'ONBOARD_SD_CS_PIN' was not declared in this scope
 #define CS_LOW()  WRITE(ONBOARD_SD_CS_PIN, LOW)  /* Set OnboardSPI cs low */
                         ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = _BV32(PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:143:3: note: in expansion of macro 'CS_LOW'
   CS_LOW();   /* CS = L */
   ^~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:41:25: note: suggested alternative: 'ONBOARD_SD_SPI'
 #define CS_LOW()  WRITE(ONBOARD_SD_CS_PIN, LOW)  /* Set OnboardSPI cs low */
                         ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = _BV32(PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:143:3: note: in expansion of macro 'CS_LOW'
   CS_LOW();   /* CS = L */
   ^~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp: In function 'void power_on()':
Marlin\src\HAL\STM32F1\onboard_sd.cpp:161:13: error: 'ONBOARD_SD_CS_PIN' was not declared in this scope
   OUT_WRITE(ONBOARD_SD_CS_PIN, HIGH); /* Set CS# high */
             ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:37:55: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)         gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                       ^~
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:41:37: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)         do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                     ^~~~~~~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:161:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(ONBOARD_SD_CS_PIN, HIGH); /* Set CS# high */
   ^~~~~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:161:13: note: suggested alternative: 'ONBOARD_SD_SPI'
   OUT_WRITE(ONBOARD_SD_CS_PIN, HIGH); /* Set CS# high */
             ^
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:37:55: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)         gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                       ^~
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:41:37: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)         do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                     ^~~~~~~~~~~
Marlin\src\HAL\STM32F1\onboard_sd.cpp:161:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(ONBOARD_SD_CS_PIN, HIGH); /* Set CS# high */
   ^~~~~~~~~
[Config.h.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5898683/Config.h.zip)
[Config_Adv.h.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5898684/Config_Adv.h.zip)

Thank you for your support!

mks-viva, 2021-02-01:
Hi! Do you want to print directly without selecting via LCD?

HarpoDog, 2021-02-01:
>   function to use the SD Card directly without a LCD (#70)

I want to print directly from the SD Card on the motherboard. I will not have an LCD installed.Sent from Samsung Galaxy smartphone.

mks-viva, 2021-02-01:
Have use MKS TFT and MKS TFT WIIF？

HarpoDog, 2021-02-01:
I do not have or plan to use  a MKS TFT or MKS TFT wifi. I need to and want to print directly from the Micro SD in the Micro SD Card slot onboard the MKS Robin  E3's motherboard by executing the g-code file to print using the Marlin Autostart feature.Are you familiar with the Marlin Autostart feature? It works very well.Sent from Samsung Galaxy smartphone.

mks-viva, 2021-02-01:
You can rename gcode file name to auto.g, and it is will auto printing when you power on.

HarpoDog, 2021-02-01:
Yes. I have always used auto0.g. IT is a really nice feature when you don't have or don't want to use a LCD.Thank you for your help so far. I look forward to your solution that will modify the Marlin firmware, allow it to complie, and enable the Autostart feature.Sent from Samsung Galaxy smartphone.

mks-viva, 2021-02-04:
At present, auto0.g is the best choice

HarpoDog, 2021-02-04:
Ok, so how do I resolve my issue above  as noted in the error message above so The MKS robin E3 with Marlin 2.0 will recognize the the Micro SD card onboard the MSK motherboard and I can print directly from the Micro SD card without an LCD or TFT? . 
How do I declare the ONBOARD_SD_CS_PIN within the scope noted and resolve the error above.
Thank you your support.

HarpoDog, 2021-02-04:
As a reminder, the errors occur when I am trying to compile the latest version of Marlin 2.0.

mks-viva, 2021-02-04:
MKS Robin E3/E3D's TF card is onboard default.

HarpoDog, 2021-02-04:
Is what you are saying is that he TF Card on the LCD or TFT is the default?
in the Marlin configuration_adv.h file - the following are the three options

 * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board.
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  //#define SDCARD_CONNECTION ONBOARD

  // Enable if SD detect is rendered useless (e.g., by using an SD extender)
  //#define NO_SD_DETECT

#endif // SDSUPPORT

When you uncomment the  //#define SDCARD_CONNECTION ONBOARD so it reads  #define SDCARD_CONNECTION ONBOARD, then compile, you get the error message above.  

This is the problem I am trying to resolve and would appreciate your help to resolve the issue.

mks-viva, 2021-02-04:
ONBOARD_SD_SS_PIN = SD_SS_PIN

HarpoDog, 2021-02-04:
Is ONBOARD_SD_SS_PIN = SD_SS_PIN what I add to Marline and if so, where do I
add this within Marlin - to which files, where within the files?

I am not sure what you are telling me below. I am not an experienced coder
so will need a little more explanation please.

Thanks

 

<https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/70#issuecomment-77
H6PLANCNFSM4W2UUVOQ> .
<https://github.com/notifications/beacon/ASVG3TEG2XZWULPSHX7S44TS5H6PLA5CNFS
M4W2UUVO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFYJJJMA
.gif> 

 

  _____  

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient> 

Scanned by McAfee
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient>  and confirmed virus-free.

mks-viva, 2021-02-05:
You need add "#define ONBOARD_SD_SS_PIN  SD_SS_PIN" in pins_MKS_ROBIN_E3.h file

mks-viva, 2021-02-08:
If have not response, I will close this issues temporaril.

closed 2021-02-08 by mks-viva

HarpoDog, 2021-02-08:
Sorry for the slow response. I work on a ship and can be days without being connected. 

I believe what I need to know is how to is how to define “ONBOARD_SD_CS_PIN” in three different scopes within the exact locations of the “Marlin\src\HAL\STM32F1\onboard_sd.cpp" file.

Your suggestion of "You need add "#define ONBOARD_SD_SS_PIN SD_SS_PIN" in pins_MKS_ROBIN_E3.h file" did not change the three error messages when compiling. 

Attached is the Marlin folder I am testing. It would be appreciated if you could test this folder,make the corrections as need to get it to compile, and send me the corrected files.

[Marlin-bugfix-2.0.x onboard test.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/5945094/Marlin-bugfix-2.0.x.onboard.test.zip)

Thank you for your support!

sn9s, 2021-02-15:
Hi, I just bought a mks robin e3d v1.1 , I have the same error when compiling.
I use the version of marlin provided by mks : https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1/Marlin-2.0.6.1_for_ender3_TMC2209

I just changed in Configuration.h :
_//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define CR10_STOCKDISPLAY_

and in Configuration_adv.h :
_#define SDCARD_CONNECTION ONBOARD_

I get the "ONBOARD_SD_CS_PIN not declared" error when changing sd connection from lcd to onboard.

When I add _#define ONBOARD_SD_SS_PIN SD_SS_PIN_ in _pins_MKS_ROBIN_E3.h_ file I get the same error.
I tried _#define ONBOARD_SD_CS_PIN SD_SS_PIN_ , it compile but when updating it get stuck at 100%.