makerbase-mks/MKS-Robin-E3-E3D #73: Robin E3 boot Ender 3 Pro

TheKillerOfMarsh, 2021-02-04:
Good evening, let's see if you can help me, I have compiled 2.0.6.1 for my Mks Robin E3 v.  1.0 correctly without errors, I just needed to change the extruder steps for my Ender 3 Pro and it updates correctly but it hangs on the Marlin logo, I have seen a .hex here is for that type of hang and where is the hex put?  Greetings.

mks-viva, 2021-02-05:
Hi! I has check marlin config about ender 3 pro, Link:
https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20Pro/CrealityV1/Configuration.h
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 93 }
[image]

TheKillerOfMarsh, 2021-02-05:
Hi! My problem is not the configuration of the extruder steps, the problem is that after compiling the Marlin downloaded from here, it does the update well and it hangs on the Marlin logo, and I only modify the steps and do not touch anything else, in  change the .bin also downloaded from here is perfect.  I don't know what can change?

mks-viva, 2021-02-06:
You mean hex file?

TheKillerOfMarsh, 2021-02-06:
Yes, I mean that when updating well and restarting it remains frozen in the Marlin logo, if with that hex file it could be solved?  incidentally, how is the hex file used?  greetings and thanks!

mks-viva, 2021-02-08:
This is not helpful, because the same marlin firmware settings, bin and hex are the same

mks-viva, 2021-02-08:
So, I will close this issues

closed 2021-02-08 by mks-viva