makerbase-mks/MKS-Robin-E3-E3D #74: I need firmware ".bin" file for ENDER 5

azatalp, 2021-02-06:
I'try to do any kind of think for setup on card. But whatever I try cant setup on mks e3 card. Every time gave me error. How 2.0.6.1 is freeze on screen after %100. Please just give working 2.0.6 fw .bin file.

azatalp, 2021-02-08:
Why anyone doesn't give me any kind of help??

mks-viva, 2021-02-08:
Hi！I am sorry！We have not add bin file for ender5，we  will add it after february 24

azatalp, 2021-02-08:
Ok I will wait for it

arsene-r2, 2021-02-28:
Hello, any updates on this?

azatalp, 2021-03-03:
I'm still waiting for this..

azatalp, 2021-03-03:
> Hi！I am sorry！We have not add bin file for ender5，we will add it after february 24

24 feb is passed? Please help us.

mks-viva, 2021-03-06:
I has upload it, you can test it, if have bug and i will fix it

mks-viva, 2021-03-06:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1

mks-viva, 2021-04-01:
I will close this issues temporily

closed 2021-04-01 by mks-viva