makerbase-mks/MKS-Robin-E3-E3D #77: SD not working and firmware update

vinacc, 2021-02-15:
I just got the Robin E3 card with firmware 2.0.6. Mounted on my Ender 3 I cannot update the firmware. Inserting the bin file inside the SD gives me the SD ERR message. How can I solve?

[image]

mks-viva, 2021-02-21:
Hi! You can try to format the TF card, and the format is FAT32

vinacc, 2021-02-21:
Hi, yes I tried to reformat it in FAT32 format but it doesn't work. I have installed OctoPi and connected a Raspberry to my Ender 3. When I start OctoPi the SD inside the Robin E3 is seen and I can also read and write but when I disconnect the raspberry the SD is no longer seen.

mks-viva, 2021-02-22:
Can you try to re-insert the card or replace the TF card to test. If this error occurs, the card status is not detected

vinacc, 2021-02-22:
already tried, the board is enabled only after connecting the raspberry. And unfortunately I am unable to perform a software update

mks-viva, 2021-02-25:
I suggest you use a new TF card

vinacc, 2021-02-28:
Through a software such as Repetier Host, by connecting the card to my Mac and sending the M21 command to the printer, the SD is mounted and viewed correctly. The problem is that turning off the printer loses the setting. It is as if the SD on the firmware were disabled by default. Is it possible to update the firmware without using the SD?

mks-viva, 2021-03-06:
Can you provide your firmware to me and i will test it?

vinacc, 2021-03-13:
> Can you provide your firmware to me and i will test it?

[Robin_e3.bin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6134898/Robin_e3.bin.zip)

mks-viva, 2021-08-19:
I'm sorry, I forgot to reply you, your firmware I updated without any problems, maybe you need to format the TF card, or replace the TF card.

closed 2021-08-23 by mks-viva