makerbase-mks/MKS-Robin-E3-E3D #79: Build Error since latest Visual Studio update

TxasMan3D, 2021-02-17:
Hello, i'm trying to build "Marling-bugfix-2.0.x" firmware with Visual Studio. This firmware was succesfully built an used during last year. The issue appears with the latest update of Visual Studio software. Maybe i have to modify or include something but i dont know what to do. Need help. 
Thank you!

[image]

TxasMan3D, 2021-02-17:
Also have the same issue with the latest repository firmwares "Marlin-2.0.6.1_for_ender3_TMC2209...."

mks-viva, 2021-02-21:
May be use the latest version marlin

TxasMan3D, 2021-02-22:
Also have the same issue using the Marlin 2.0.7.2 configuration files...

mks-viva, 2021-02-25:
I will test the latest version marlin

mks-viva, 2021-02-25:
[image]
I test it is OK, My firmware
[Marlin-bugfix-2.0.x.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6040334/Marlin-bugfix-2.0.x.zip)

closed 2021-04-24 by mks-viva