makerbase-mks/MKS-Robin-E3-E3D #80: V1.0 and V1.1

ll777, 2021-02-17:
What is the difference between the V1.0 and V1.1 Firmware? Their description is the same:
V1.0 - Upload E3/E3D V1.1 firmware
V1.1 - Upload E3/E3D V1.1 firmware

mks-viva, 2021-02-20:
Hi, yon? can refer to hardware path information

ll777, 2021-02-20:
The Full path for them is:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware
where I can find both folders 
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.0
and
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1

mks-viva, 2021-02-21:
Like this: 
[image]

closed 2021-02-21 by mks-viva

Amppa, 2021-03-27:
@mks-viva 
So if the PCB print with "MKS Robin E3 V1.0", means it is V1.0 hardware right?
[image]

I noticed that V1.0 used Marlin ver2.0.6. The V1.1 is built from Marlin ver2.0.6.1
I wonder if I can build V1.0 on the newest Marlin

mks-viva, 2021-03-27:
> @mks-viva
> So if the PCB print with "MKS Robin E3 V1.0", means it is V1.0 hardware right?
> [image]
> 
> I noticed that V1.0 used Marlin ver2.0.6. The V1.1 is built from Marlin ver2.0.6.1
> I wonder if I can build V1.0 on the newest Marlin

Yes，you can. You only define motherboard is MKS_ROBIN_E3_V1_0